fivdi/i2c-bus #5: What i should use as command code inside readI2cBlockSync?

simar88, 2015-04-13:
Hi @fivdi , i'm new about using i2c modules and i'm playing around with my new raspberry pi 2 and hover  board from hoverlabs.
I'm trying to translate a python library, that you can view <a href="https://github.com/jonco91/hover_raspberrypi/blob/master/Hover_library.py">here</a>, in a node.js modules to use inside a project but i don't know what i have to use for this parameter.
Can you say me something about it?
Thanks in advance,
Simone

fivdi, 2015-04-13:
In general, the data sheets for I2C devices describe a register map or a command set which can be used to access the device. In reality, it's the same thing. For example, one temperature sensor could have a register called TEMP and another temperature sensor could have a command called ACCESS_TEMP. In the first sensor, the TEMP register is accessed to get the temperature. In the second sensor, the ACCESS_TEMP command is performed to get the temperature. The same thing is happening in each case.

The following Python [line of code](https://github.com/jonco91/hover_raspberrypi/blob/master/Hover_library.py#L79) from the Hover library:
```python
busData = bus.read_i2c_block_data(self.address,0,18)
```
would translate to something like this in JavaScript:
```JavaScript
var buf = new Buffer(18);
bus.readI2cBlockSync(this.address, 0, 18, buffer);
```
In this case the command or register is 0, and 18 bytes of data are being read.

simar88, 2015-04-14:
Thanks @fivdi for the fast reply and for the explanation.
I'll try your suggestion in these day.
Thanks again! :)
Simone

fivdi, 2015-04-14:
No problem and good luck with Hover. I guess the issue can be closed so. Feel free to reopen if Hover can't be accessed with i2c-bus.

closed 2015-04-14 by fivdi