fivdi/i2c-bus #6: node-i2c .writeBytes(cmd, bytes) equivalent with i2c-bus ?

Stead68, 2015-05-05:
Hello,

I totally rewrote adafruit-i2c-lcd which works with node-i2c, because it seems to be broken with latest version of node.js.
I decided to use i2c-bus, but I don't found equivalent of wire.writeBytes(cmd, bytes);
I tried:

var buf = new Buffer(values);
bus.writeI2cBlockSync(this.ADDRESS, cmd, buf.length, buf);

But data length are over 32 bytes and it doesn't works.
Do you have any idea about how can I fix it ?

Thanks guys.

fivdi, 2015-05-06:
i2c-bus writeI2cBlockSync should work as a replacement for node-i2c writeBytes,

Under the covers, node-i2c writeBytes has the same 32 byte limitation as can be seen [here](https://github.com/kelly/node-i2c/blob/master/src/i2c-dev.h#L306-L307), but it just ignores the 33rd byte and beyond. i2c-bus writeI2cBlockSync [throws an error](https://github.com/fivdi/i2c-bus/blob/master/src/writei2cblock.cc#L119-L122) when passed more than 32 bytes.

Perhaps there's a logical error in the original adafruit-i2c-lcd or in the rewrite. Maybe in [this](https://github.com/fehmer/adafruit-i2c-lcd/blob/master/src/adafruit_i2c_lcd.coffee#L234-L245) section of code which handles the 32 byte limit specially?

Stead68, 2015-05-06:
Can I give you the code I rewrote to got an external point of view, because it seems to not work as intended ?

Thanks for your reply :)

fivdi, 2015-05-06:
Sure, add the code here in the issue.

Stead68, 2015-05-06:
// Port expander registers
var MCP23017_IOCON_BANK0    = 0x0A; // #IOCON when Bank 0 active
var MCP23017_IOCON_BANK1    = 0x15; // #IOCON when Bank 1 active
// These are register addresses when in Bank 1 only:
var MCP23017_GPIOA          = 0x09;
var MCP23017_IODIRB         = 0x10;
var MCP23017_GPIOB          = 0x19;

// LCD Commands
var LCD_CLEARDISPLAY        = 0x01;
var LCD_RETURNHOME          = 0x02;
var LCD_ENTRYMODESET        = 0x04;
var LCD_DISPLAYCONTROL      = 0x08;
var LCD_CURSORSHIFT         = 0x10;
var LCD_FUNCTIONSET         = 0x20;
var LCD_SETCGRAMADDR        = 0x40;
var LCD_SETDDRAMADDR        = 0x80;

// Flags for display on/off control
var LCD_DISPLAYON           = 0x04;
var LCD_DISPLAYOFF          = 0x00;
var LCD_CURSORON            = 0x02;
var LCD_CURSOROFF           = 0x00;
var LCD_BLINKON             = 0x01;
var LCD_BLINKOFF            = 0x00;

// Flags for display entry mode
var LCD_ENTRYRIGHT          = 0x00;
var LCD_ENTRYLEFT           = 0x02;
var LCD_ENTRYSHIFTINCREMENT = 0x01;
var LCD_ENTRYSHIFTDECREMENT = 0x00;

// Flags for display/cursor shift
var LCD_DISPLAYMOVE         = 0x08;
var LCD_CURSORMOVE          = 0x00;
var LCD_MOVERIGHT           = 0x04;
var LCD_MOVELEFT            = 0x00;

var flip = [ 0x00, 0x10, 0x08, 0x18,
		     0x04, 0x14, 0x0C, 0x1C,
		     0x02, 0x12, 0x0A, 0x1A,
		     0x06, 0x16, 0x0E, 0x1E ];

var pollables = [ LCD_CLEARDISPLAY, LCD_RETURNHOME ];

var EventEmitter = require('events').EventEmitter;
var I2C = require('i2c-bus');

var Plate = function(device, address, pollInterval) {

	this.ADDRESS = address;
	this.PORTA   = 0;
	this.PORTB   = 0;
	this.DDRB    = 0x10;
	this.WIRE    = I2C.openSync(1);

	this.BSTATE = 0;

	this.colors = {
		OFF: 0x00,
		RED: 0x01,
		GREEEN: 0x02,
		YELLOW: 0x03,
		BLUE: 0x04,
		VIOLET: 0x05,
		TEAL: 0x06,
		WHITE: 0x07,
		ON: 0x07
	}

[... 193 lines cut ...]
	/*if (pollInterval > 0)
	{
		this.poll = setInterval(function() {
			var cur = this.buttonState();

			if (cur != this.BSTATE)
			{
				var key = this.BSTATE ^ cur;

				emit("button_change", key);

				if (cur < this.BSTATE)
					emit("button_up", key);
				else
					emit("button_down", key);

				this.BSTATE = cur;
			}
		}, pollInterval);
	}*/

	this.init();
}

module.exports = Plate;

Stead68, 2015-05-06:
I tried with writeI2cBlockSync in sendBytes method but it throw an error.
I tried to translate adafruit_i2c_lcd.coffee to javascript.

fivdi, 2015-05-06:
I can't see any calls to writeI2cBlockSync in the above code but see that sendBytes is now using i2cWriteSync. What currently doesn't function correctly?

Stead68, 2015-05-06:
// Port expander registers
var MCP23017_IOCON_BANK0    = 0x0A; // #IOCON when Bank 0 active
var MCP23017_IOCON_BANK1    = 0x15; // #IOCON when Bank 1 active
// These are register addresses when in Bank 1 only:
var MCP23017_GPIOA          = 0x09;
var MCP23017_IODIRB         = 0x10;
var MCP23017_GPIOB          = 0x19;

// LCD Commands
var LCD_CLEARDISPLAY        = 0x01;
var LCD_RETURNHOME          = 0x02;
var LCD_ENTRYMODESET        = 0x04;
var LCD_DISPLAYCONTROL      = 0x08;
var LCD_CURSORSHIFT         = 0x10;
var LCD_FUNCTIONSET         = 0x20;
var LCD_SETCGRAMADDR        = 0x40;
var LCD_SETDDRAMADDR        = 0x80;

// Flags for display on/off control
var LCD_DISPLAYON           = 0x04;
var LCD_DISPLAYOFF          = 0x00;
var LCD_CURSORON            = 0x02;
var LCD_CURSOROFF           = 0x00;
var LCD_BLINKON             = 0x01;
var LCD_BLINKOFF            = 0x00;

// Flags for display entry mode
var LCD_ENTRYRIGHT          = 0x00;
var LCD_ENTRYLEFT           = 0x02;
var LCD_ENTRYSHIFTINCREMENT = 0x01;
var LCD_ENTRYSHIFTDECREMENT = 0x00;

// Flags for display/cursor shift
var LCD_DISPLAYMOVE         = 0x08;
var LCD_CURSORMOVE          = 0x00;
var LCD_MOVERIGHT           = 0x04;
var LCD_MOVELEFT            = 0x00;

var flip = [ 0x00, 0x10, 0x08, 0x18,
		     0x04, 0x14, 0x0C, 0x1C,
		     0x02, 0x12, 0x0A, 0x1A,
		     0x06, 0x16, 0x0E, 0x1E ];

var pollables = [ LCD_CLEARDISPLAY, LCD_RETURNHOME ];

var EventEmitter = require('events').EventEmitter;
var I2C = require('i2c-bus');

var Plate = function(device, address, pollInterval) {

	this.ADDRESS = address;
	this.PORTA   = 0;
	this.PORTB   = 0;
	this.DDRB    = 0x10;
	this.WIRE    = I2C.openSync(1);

	this.BSTATE = 0;

	this.colors = {
		OFF: 0x00,
		RED: 0x01,
		GREEEN: 0x02,
		YELLOW: 0x03,
		BLUE: 0x04,
		VIOLET: 0x05,
		TEAL: 0x06,
		WHITE: 0x07,
		ON: 0x07
	}

[... 210 lines cut ...]
			}
		}, pollInterval);
	}*/

	this.init();
}

module.exports = Plate;

----------------------------------------

I replaced "for (var k = 0; k <= last; k++)" by "for (var k = 0; k < last; k++)"
It seems to work without throw exception.

Then on my main program I use:
var LCDPlate = require("./LCD.js");
var LCD = new LCDPlate('/dev/i2c-1', 0x20);

LCD.backlight(th.colors.red);
LCD.message("Hello world!");

On my PI the LCD turn on for 1 second and all LEDs are turning off and a special char is displayed.
LEDs never turn red, and hello world is never displayed.

Thanks for help :-)

fivdi, 2015-05-06:
Hmm, ...

Sorry, but I'm afraid I can't help you out here. There's too much code there for attempting to figure out what's going wrong.

fivdi, 2015-05-17:
@Stead68 I'm going to close this one. If there's a concrete issue with i2c-bus, feel free to re-open or create a new issue.

closed 2015-05-17 by fivdi