fivdi/i2c-bus #10: Deprecate names containing substring "i2c"

fivdi, 2015-09-28:
old name | new name
:--- | :---
i2cFuncs | funcs
i2cFuncsSync | funcsSync
i2cRead | read
i2cReadSync | readSync
i2cWrite | write
i2cWriteSync | writeSync
readI2cBlock | readBlock
readI2cBlockSync | readBlockSync
writeI2cBlock | writeBlock
writeI2cBlockSync | writeBlockSync
I2cFuncs | funcs

clinklet, 2016-01-10:
With the readi2cBlock deprecated, how am I supposed to read 3 (or 4, depending on settings) bytes from my adc device?  Not to mention that I am getting the error 
node: ../src/node_buffer.cc:103: char* node::Buffer::Data(v8::Handle<v8::Object>): Assertion `obj->HasIndexedPropertiesInExternalArrayData()' failed.
when running it.

fivdi, 2016-01-10:
The issue is poorly titled. Nothing will be removed. However, the sub-string `i2c` will be removed from names as it's not needed. For example, `readi2cBlock` will be renamed to `readBlock`.

Regarding the error you're seeing, the tests for readi2cBlock are passing at the moment and the code in this [gist](https://gist.github.com/fivdi/190d0849d75de41a87e4) calls readi2cBlock to read 6 bytes from an ADXL345 successfully.

If there are still problems, please create a separate issue for handling the topic. Please provide more information there, for example, platform, operating system, version of Node.js, version of i2c-bus, test code to reproduce the error, complete stack trace, ...

fivdi, 2016-02-13:
On second thoughts, the proposed renaming the isn't actually a good idea.

The following four method name changes are not a good idea as the methods are for plain I2C and are unrelated to the SMBUS specification. The i2c prefix is therefore a good idea rather than a bad one.

old name | new name
:--- | :---
i2cRead | read
i2cReadSync | readSync
i2cWrite | write
i2cWriteSync | writeSync

The following four method name changes are not a good idea as the methods are also unrelated to the SMBUS specification. In addition, there are already undocumented methods called readBlock, readBlockSync, writeBlock, and writeBlockSync that are related to the SMBUS specification. They haven't been documented yet due to lack of hardware for testing.

old name | new name
:--- | :---
readI2cBlock | readBlock
readI2cBlockSync | readBlockSync
writeI2cBlock | writeBlock
writeI2cBlockSync | writeBlockSync

The leaves us with three remaining proposed changes. Two are method names and one is a constructor function name. They are related to both the SMBUS protocol and I2C so the current names are reasonable and shouldn't result in any confusion.

old name | new name
:--- | :---
i2cFuncs | funcs
i2cFuncsSync | funcsSync
I2cFuncs | funcs

closed 2016-02-13 by fivdi