fivdi/i2c-bus #14: Problem running on node 4.2.1 and debian Jessie

teuteuguy, 2015-10-19:
When trying to the run a simple code I get stuck with it failing.
Any ideas would be welcome.

So far this is my code

var i2c = require('i2c-bus');

var HTS221_ADDR = 0xBF;
var CMD_TEMP_HIGH = 0x2B;
var CMD_TEMP_LOW = 0x2A;
var CMD_HUMID_HIGH = 0x29;
var CMD_HUMID_LOW = 0x28;

module.exports.get = function(cb) {

  var i2c1 = i2c.openSync(1);

  var rawTemp = i2c1.readByteSync(HTS221_ADDR, CMD_TEMP_HIGH);
  console.log(rawTemp);

  i2c1.closeSync();

};

_________________

i2c.setAddrSync(peripheral, addr);
                                             ^

Error: EINVAL, Invalid argument
    at Error (native)
    at peripheralSync (/home/pi/dev/node/node_modules/raspberry-sensor-sense-hat/node_modules/i2c-bus/i2c-bus.js:89:46)
    at Bus.readByteSync (/home/pi/dev/node/node_modules/raspberry-sensor-sense-hat/node_modules/i2c-bus/i2c-bus.js:162:27)
    at Object.module.exports.get (/home/pi/dev/node/node_modules/raspberry-sensor-sense-hat/index.js:13:22)
    at Object.<anonymous> (/home/pi/dev/node/app.js:8:7)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Function.Module.runMain (module.js:467:10)

fivdi, 2015-10-19:
Please use the i2cdetect to determine whether or not there really is a device at address 0xbf and report your findings. This can be done with the following command:

```
i2cdetect -y -r 1
```

if it's not already installed, i2cdetect can be installed with `sudo apt-get install i2c-tools` (I think)

teuteuguy, 2015-10-19:
Thanks for this great tool !
This is what I get from the response.

```
i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- UU -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- 5c -- -- 5f
60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Basically I'm trying to interact with the sense-hat. Reading up on it, I got the 0xBF from the HTS221 humidity sensor.

fivdi, 2015-10-19:
Although the [documentation for the HTS221](http://www.st.com/web/en/resource/technical/document/datasheet/DM00116291.pdf) speaks of 8-bit addresses, they really only have 7 bits. Bit 0 (the least significant bit) shouldn't be taken into account. The correct address in this case is probably 0x5f. Please try with address 0x5f to see if it works.

teuteuguy, 2015-10-19:
OK, my bad. Sorry about that. Works perfect on 0x5f.
I'll close the "issue".
Cheers

closed 2015-10-19 by teuteuguy

fivdi, 2015-10-19:
That's good news. Have fun with i2c-bus :)

teuteuguy, 2015-10-19:
Thanks !
CHeers