fivdi/i2c-bus #16: Interrupt based callback?

crosofg, 2016-01-18:
I am trying to use this for MCP23017. How do I set a interrupt based callback. I need to get the onchange callback .

fivdi, 2016-01-18:
If I'm not mistaken, the interrupt-on-change interrupts can be detected on the INTA/INTB pins of the MCP23017. Is this correct? If so, these pins could be wired to GPIO pins configured as inputs on the Raspberry Pi / BeagleBone, or whatever the system happend to be. The [onoff](https://github.com/fivdi) package can then be used to detect these interrupts. onoff can be used for low latency interrupt detection (without polling.)

fivdi, 2016-01-18:
Please let me know whether or not I've interpreted the requirement correctly here.

crosofg, 2016-01-18:
Yes, you are correct. Interrupt are detected on INTA / B. I think onff package can work

fivdi, 2016-01-18:
Ok, give it a try and let us know how it goes :)

fivdi, 2016-01-22:
@crosofg was it possible to resolve the issue with onoff?

crosofg, 2016-01-23:
@fivdi yes that worked. Thanks

closed 2016-01-23 by crosofg

fivdi, 2016-01-23:
Excellent, thanks for the feedback.