fivdi/i2c-bus #17: Installation Issue

crosofg, 2016-01-27:
Hi
I am trying to install i2c-bus module. But I am getting following error
```
> i2c-bus@1.0.1 install /home/pi/Desktop/gg/node_modules/i2c-bus
> node-gyp rebuild

gyp ERR! Completion callback never invoked! 
gyp ERR! System Linux 4.1.14-v7+
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/Desktop/gg/node_modules/i2c-bus
gyp ERR! node -v v0.10.0
gyp ERR! node-gyp -v v3.2.1
gyp ERR! This is a bug in `node-gyp`.
gyp ERR! Try to update node-gyp and file an Issue if it does not help:
gyp ERR!     <https://github.com/nodejs/node-gyp/issues>
npm WARN enoent ENOENT, open '/home/pi/Desktop/gg/package.json'
npm WARN gg No description
npm WARN gg No repository field.
npm WARN gg No README data
npm WARN gg No license field.
npm ERR! Linux 4.1.14-v7+
npm ERR! argv "node" "/usr/local/bin/npm" "install" "i2c-bus"
npm ERR! node v0.10.0
npm ERR! npm  v3.5.3
npm ERR! code ELIFECYCLE

npm ERR! i2c-bus@1.0.1 install: `node-gyp rebuild`
npm ERR! Exit status 6
npm ERR! 
npm ERR! Failed at the i2c-bus@1.0.1 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the i2c-bus package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs i2c-bus
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls i2c-bus
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:

```

Wingpad, 2016-01-27:
I'm having this issue as well - running on node v0.10.29, npm 1.4.21 and g++ v4.9.2 on Raspbian Jessie.

fivdi, 2016-01-27:
Node.js v0.10.0, which was the version used during the above install, is very old and no longer supported by [nan](https://github.com/nodejs/nan). i2c-bus depends on nan. nan does support Node.js v0.10, but only v0.10.29 or higher. It looks like the install was performed on a Raspberry Pi 2. The latest version of Raspbian Jessie comes with Node.js v0.10.29 pre-installed, but unfortunately it's a broken version of Node.js. See [here](https://github.com/fivdi/onoff/wiki/Node.js-v0.10.29-and-native-addons-on-the-Raspberry-Pi) for further information.

The Node.js Foundation provides binary distributions of Node.js v4.0.0 and higher for the Rapsberry Pi 1 and 2. These binary distributions can be found in the correspondingly named directory [here](https://nodejs.org/download/release/). For example, the binaries for Node.js v4.2.6 are [here](https://nodejs.org/download/release/v4.2.6/). The armv6l distribution is for the Raspberry Pi 1 and the armv7l distribution is for the Raspberry Pi 2.

I'd suggest using a newer version of Node.js, for example, v4.2.6.

crosofg, 2016-01-31:
Thanks. it helped.

closed 2016-01-31 by crosofg

fivdi, 2016-01-31:
That's good to hear :)