fivdi/i2c-bus #18: writeFailed Rasp -> Atmega

bidhouilles, 2016-03-20:
Hello,

First, thanks for your works, it's very good work, really.
I turned around this problem for couple of days and I try to find a forum to post this message instead of opening an issue but I didn't find any Forum !!!

I have a raspberry connected to a Atmega328P (Arduino on breaboard). I can communicate with I2c with ReadAsync (rasp -> Atmega) without problem.

But when I want to use writeByteSync (or any write function) in nodejs and I have ETIMEOUT. Unfortunately, the I2C interface Rasp is OUT and impossible to use ( search addresse become very slow :i2cdetect -y 1) after executing the code. I must reboot the Rasp. I have the feeling that an issue broke the interface.

When I use :i2cset -y 1 0x08 0x01 0x01 it's working fine my Atmega get the msg.

code:
var i2c = require('i2c-bus'),
  i2c1 = i2c.openSync(1);
// Enter one shot mode (this is a non volatile setting)
  i2c1.writeByteSync(0x08, 0x01, 0x01);  -> ETIMOUT
 i2c1.closeSync();

thx,

David

fivdi, 2016-03-20:
>First, thanks for your works, it's very good work, really.

Thanks :)

If `i2cset -y 1 0x08 0x01 0x01` functions correctly but the code posted above doesn't then I'm at a bit of a loss and not sure if I can help any further.

Was the test with `i2cset` and the code above performed with exactly the same hardware, i.e., exactly the same Raspberry Pi and exactly the same ATmega328P?

When executed, the command `i2cset -y 1 0x08 0x01 0x01` calls the Linux `i2c_smbus_read_byte_data` function internally. The i2c-bus `writeByteSync` method also calls `i2c_smbus_read_byte_data` internally.

The [avr-io](https://github.com/fivdi/avr-io) repository is an ATmega328P I2C project. If the AVR program from that project is flashed onto an ATmega328P, the following code can be used to blink an LED connected to PB0 on the AVR from the Raspberry Pi without a problem (I just tried it):

```js
var i2c = require('i2c-bus').openSync(1);

var OUTPUT = 1;

var AVR_ADDR = 0x28,
  P8_MODE = 0x20,
  P8_VALUE = 0x21;

i2c.writeByteSync(AVR_ADDR, P8_MODE, OUTPUT);

setInterval(function () {
  i2c.writeByteSync(AVR_ADDR, P8_VALUE,
    i2c.readByteSync(AVR_ADDR, P8_VALUE) ^ 1);
}, 500);
```

Note that the Raspberry Pi has a hardware bug that prevents it from supporting I2C clock stretching correctly. The avr-io project was implemented with this bug in mind. On an ATmega328P running at 3.3V/8MHz, the avr-io project can handle I2C clock speeds of up to 100KHz without clock stretching.

Has your project also taken the clock stretching bug on the Raspberry Pi into account? If not, you could reduce the I2C clock speed to say 10KHz to see if it functions (see [here](https://github.com/fivdi/i2c-bus/blob/master/doc/raspberry-pi-i2c.md#step-3---setting-the-i2c-baudrate) for details).

bidhouilles, 2016-03-20:
Thanks for reactivity on a Sunday . 
I used i2cset  to be sure that hardware work fine with linux API. and to be sure (or not) the hardware interface was KO (or not) after ETIMOUT happened.
So after lunch I made a mistake and deleted all my files and modules... But, the good thing : I had to reinstall all of them. What a big surprise :+1: everything I wrote work ! Maybe the fact to reinstall all the package made it work !!!
Because I'm sure that something didn't work.
I will try to reduce clock to 10kHz, just in case...

Thx again, I had to go, my rasp need me...

David.

closed 2016-03-20 by bidhouilles