fivdi/i2c-bus #19: Use of CMD parameter

shekit, 2016-04-23:
Hi,

I had a few questions which are unclear from the documentation. I was hoping you could help clarify.

1. What is the command parameter used for in functions such as `bus.writeByte(addr, cmd, byte, cb)`
2. What is the difference between sendByte and writeByte
3. What is the difference between plain i2c and SMBus?

bidhouilles, 2016-04-23:
Hello,
I'm not an expert but I have a raspberry which communicate with Atmega with I2C (sync and async way), doest it make me an expert ? :-)1. I do not use this api but I can tell you what I understood in the exmple : you can pass a cmd and parameter (byte) specific to your device2. Good question ! I use only I2CWrite and I2CReadSync. Sorry... 3. SMB is an evolution of the I2C, many device are I2C but not SMB. You can find API (don't remember where) which API compatible on your device.
I hope I could have help you.
Thx
David.

fivdi, 2016-04-23:
> What is the command parameter used for in functions such as bus.writeByte(addr, cmd, byte, cb)

This question is similar to the question asked in #15. The answer can be found there.

> What is the difference between sendByte and writeByte

See section `2.3.2 Send byte` and section `2.3.4 Write byte/word` of [AVR316](http://www.atmel.com/Images/doc2583.pdf).

> What is the difference between plain i2c and SMBus?

For some information about the differences between I2C and SMBus see section `2.1 Differences between SMBus and I2C` of [AVR316](http://www.atmel.com/Images/doc2583.pdf). Unlike the majority of the SMBus methods the plain I2C methods don't have a `cmd` parameter. These plain I2C methods can be used to send a block of data to the I2C device without specifying a `cmd`.

closed 2016-04-25 by fivdi