fivdi/i2c-bus #21: Problem installing on node 0.10.38

colinbes, 2016-05-25:
I am attempting to install i2c-bus on beaglebone black with nodejs v0.10.38 installed. At this stage, I am not able to update nodejs due to product life-cycle.

When running npm install i2c-bus I get the error (as stated in readme) even though nodejs should be working solution.

```
> i2c-bus@1.1.0 install /opt/project/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory `/opt/project/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
In file included from ../src/i2c.cc:2:0:
../node_modules/nan/nan.h:330:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
make: *** [Release/obj.target/i2c/src/i2c.o] Error 1
make: Leaving directory `/opt/project/node_modules/i2c-bus/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:12)
gyp ERR! System Linux 3.8.13-bone68b
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /opt/project/node_modules/i2c-bus
gyp ERR! node -v v0.10.38
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
```
 Note that I am running node v0.10.38 which is stated should work.

fivdi, 2016-05-25:
Let's check to see if the version of Node.js v0.10.38 that's being used contains the required UTF8 patch. Edit `v8.h` which is likely to be located at `/usr/include/node/v8.h` on a standard Debian distro for the BeagleBone. Then search for the `WriteOptions' enum.

If `WriteOptions` has the following 5 options, then required patch is present and everything should actually work:
```cpp
  enum WriteOptions {
    NO_OPTIONS = 0,
    HINT_MANY_WRITES_EXPECTED = 1,
    NO_NULL_TERMINATION = 2,
    PRESERVE_ASCII_NULL = 4,
    REPLACE_INVALID_UTF8 = 0 // there may be an 8 here
  };
```
If WriteOptions has the following 4 options, then the required patch is not there:
```
  enum WriteOptions {
    NO_OPTIONS = 0,
    HINT_MANY_WRITES_EXPECTED = 1,
    NO_NULL_TERMINATION = 2,
    PRESERVE_ASCII_NULL = 4,
  };
```
If `REPLACE_INVALID_UTF8 = 0` is missing, add it, and then check to see if i2c-bus can be installed.

colinbes, 2016-05-25:
Just to be sure, again confirmed `node --version` shows v0.10.38

On my beaglebone, the v8.h file is at `/usr/include/nodejs/deps/v8/include` and doesn't have the REPLACE_INVALID_UTF8 entry in it.

```
enum WriteOptions {
    NO_OPTIONS = 0,
    HINT_MANY_WRITES_EXPECTED = 1,
    NO_NULL_TERMINATION = 2,
    PRESERVE_ASCII_NULL = 4
};
```

Adding `REPLACE_INVALID_UTF8 = 0` to WriteOptions allows i2c-bus to be installed.

Thanks for help. Sad it takes a patch to v8.h file as it's not ideal having non-standard release installed in machine, but it's a way forward.

~C

closed 2016-05-25 by colinbes

colinbes, 2016-05-25:
Just want to confirm it should be REPLACE_INVALID_UTF8 = 0 and not REPLACE_INVALID_UTF8 = 8?

fivdi, 2016-05-25:
Yes, it a Debian unstable patch: https://sources.debian.net/src/libv8-3.14/3.14.5.8-10/debian/patches/nodejsREPLACE_INVALID_UTF8.patch/

It's a workaround that's disputed: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=797545

The version of Node.js on your BeagleBone is broken. If you can't upgrade to a non-broken version this is the only solution that I'm aware of.

colinbes, 2016-05-25:
Broken node.js was not expected as its untouched

Thanks again

Sent via mobile