fivdi/i2c-bus #23: 32 byte limitation

sgoll, 2016-08-03:
I need to send payloads with more than 32 bytes to a connected peripheral (using `raspi-i2c`). The peripheral speaks plain I2C, not SMBus.

Unfortunately, the I2C methods provided seem to be hardwired to use the SMBus functions, in particular they all seem to fall back to `i2c_smbus_access` in the end.

Is it somehow possible to use plain I2C without the 32 byte limitation? According to this post http://stackoverflow.com/a/25984891, the Linux kernel seems to be able to send longer I2C packets when not using SMBus.

What would be the easiest way of sending longer packets using `i2c-bus` (and `raspi-i2c`)?

fivdi, 2016-08-03:
i2c-bus supports both plain I2C and SMBus. The following four methods are plain I2C methods and should support payloads with more than 32 bytes:

[i2cRead](https://github.com/fivdi/i2c-bus#busi2creadaddr-length-buffer-cb)
[i2cReadSync](https://github.com/fivdi/i2c-bus#busi2creadsyncaddr-length-buffer)
[i2cWrite](https://github.com/fivdi/i2c-bus#busi2cwriteaddr-length-buffer-cb)
[i2cWriteSync](https://github.com/fivdi/i2c-bus#busi2cwritesyncaddr-length-buffer)

The corresponding raspi-i2c methods are:

[read](https://github.com/nebrius/raspi-i2c/blob/3.1.0/index.js#L128)
[readSync](https://github.com/nebrius/raspi-i2c/blob/3.1.0/index.js#L157)
[write](https://github.com/nebrius/raspi-i2c/blob/3.1.0/index.js#L264)
[writeSync](https://github.com/nebrius/raspi-i2c/blob/3.1.0/index.js#L285)

The four raspi-i2c methods have an optional `register` argument. If the `register` argument is not specified, the raspi-i2c methods will call the i2c-bus plain I2C methods. Unfortunately, the raspi-i2c methods explicitly limit the payload size to 32 bytes currently (see [checkLength](https://github.com/nebrius/raspi-i2c/blob/3.1.0/index.js#L139).)

I'd suggest trying with the four i2c-bus plain I2C methods to see if it works.

sgoll, 2016-08-03:
@fivdi Thank you for the detailed explanation. Somehow I was under the impression that the exception was thrown in `i2c-bus` code when I tried using the plain I2C functions provided by `raspi-i2c`.

I opened a ticket at `raspi-i2c` to discuss the issue there: nebrius/raspi-i2c#9.

closed 2016-08-03 by sgoll