fivdi/i2c-bus #24: 30 byte read limit

shtaif, 2016-08-07:
Hi,

A lot of MEMS sensors offer a FIFO buffer to allow retrieval of multiple samples in one read which reduces the bandwidth usage, right?

So to speak, this library enforces the standard allowing to read a max of 30 bytes in one operation. Regarding the FIFO buffer feature, this let's me retrieve 5 samples at a time.

However, in many cases, sensors' FIFO buffer can store up to 32 samples to read at a time (which in my case, equals 192 bytes), which can therefore greatly optimize system efficiency.

Is there any approach, with or without this library, that can be used to read more than 30 bytes sequentially, in order to e.g: utilize the FIFO buffer to the fullest?
Or perhaps, for some reason, it's better not to, and thus the standard limit exists?

Thanks!

fivdi, 2016-08-07:
i2c-bus supports both plain I2C and SMBus. The following four methods are plain I2C methods and should support payloads with more than 32 bytes:

[i2cRead](https://github.com/fivdi/i2c-bus#busi2creadaddr-length-buffer-cb)
[i2cReadSync](https://github.com/fivdi/i2c-bus#busi2creadsyncaddr-length-buffer)
[i2cWrite](https://github.com/fivdi/i2c-bus#busi2cwriteaddr-length-buffer-cb)
[i2cWriteSync](https://github.com/fivdi/i2c-bus#busi2cwritesyncaddr-length-buffer)

shtaif, 2016-08-07:
Thanks, cool.
How would one specify the cmd/address when uses the i2cRead method?

fivdi, 2016-08-07:
> How would one specify the cmd/address when uses the i2cRead method?

[This test code](https://github.com/fivdi/i2c-bus/blob/v1.1.1/test/async.js#L41-L54) shows how to do it. The trick is to use two calls. A call to `i2cWrite` to write the command followed by a call to `i2cRead` to read the results.

shtaif, 2016-08-07:
Nice, closing now...

closed 2016-08-07 by shtaif