fivdi/i2c-bus #25: w1_ds28e17 interoperarability issue when using scan or scanSync functions

gundrygs, 2016-10-05:
The DS28E17 chip from Maxim is a 1-Wire to I2C bridge - details here:
[https://www.maximintegrated.com/en/products/interface/controllers-expanders/DS28E17.html](url)
 
A kernel driver for Linux has been created by Jan Kandziora, and we are using this on Raspbian Jessie 4.4.22  with great results - details here: [https://github.com/ianka/w1_ds28e17](url)

This presents an additional i2c device in Linux, in our case /dev/i2c-3, which can be addressed using the i2c-bus library. When doing a bus scan, i2c-bus tries a I2C_FUNC_SMBUS_READ_BYTE on each slave address in turn, from 0x03 through to 0x77. The native i2c device /dev/i2c-1 replies with an EIO error code for unoccupied slave addresses, but the w1_ds28e17 replies with ENXIO, which causes the i2c-bus library to throw an exception. I have discussed this with Jan, and agree with him that this is in fact correct behaviour - see the points he makes below:

_"linux/Documentation/i2c/fault-codes says:

ENXIO
  Returned by I2C adapters to indicate that the address phase
  of a transfer didn't get an ACK.  While it might just mean
  an I2C device was temporarily not responding, usually it
  means there's nothing listening at that address.

  Returned by driver probe() methods to indicate that they
  found no device to bind to.  (ENODEV may also be used.)

So I have to assume returning ENXIO is correct behaviour. I'm pretty sure the other drivers will get patched someday.

The reason why w1_ds28e17 doesn't return EIO on I2C addressing failed is the w1 subsystem returns EIO when addressing the DS28E17 fails. If both returned the same error code, you had no chance to find out if it was the DS28E17 missing or the I2C slave."_

Would you consider amending the i2c-bus library behaviour to ignore returned ENXIO error codes in these circumstances ?

fivdi, 2016-10-05:
Thanks for reporting this issue.

i2c-bus v1.1.2 has just been published to npm and its should resolve this issue. The fix is slightly different to the fix suggested above. The scan and scanSync methods will now continue to scan on all errors.

Please give it a try and report your findings.

gundrygs, 2016-10-06:
Many thanks Brian - working fine now - happy for the issue to be closed :-)

fivdi, 2016-10-06:
Ok. That's good news.

closed 2016-10-06 by fivdi