fivdi/i2c-bus #26: Can't force I2C bus commands

tchilton, 2016-11-29:
The standard Linux i2cget and i2cset tools (from i2c-tools) includes a -f flag that allows commands to be forced. This is used when the I2C device is managed by a kernel device driver (i2cdetect shows UU for that device). This library doesn't include equivalent functionality, hence some devices are not accessible in the same manner that can be achieved from the command line.

An example use case is a real-time clock (RTC) device at 0x68 which is used by the OS RTC module, but where you might want to set an alarm or read the RTC temperature in your application. 

I have determined that this is is handled by the i2cget/i2cset commands by a change to the ioctl command as shown in line 408 of i2cbusses.c from i2c-tools. See 
https://fossies.org/dox/i2c-tools-3.1.2/i2cbusses_8c_source.html
The ioctl passes either I2C_SLAVE_FORCE or I2C_SLAVE

I have also identified that the i2c-bus library's SetAddr function in setaddr.cc only passes I2C_SLAVE, no force option exists.

Could the library be extended to include an optional force flag on the i2cWrite / i2cRead set of functions so that full functionality can be achieved ? 

I guess that the documentation should also include a suitable warning about fiddling with kernel owned devices if this option is used. The wording in the i2cset/i2cget commands is as follows :

Force access to the device even if it is already busy. By default, i2cset will refuse to access a device which is already under the control of a kernel driver. Using this flag is dangerous, it can seriously confuse the kernel driver in question. It can also cause i2cset to silently write to the wrong register. So use at your own risk and only if you know what you're doing.

fivdi, 2016-12-02:
Did you fork and modify i2c-bus to use `I2C_SLAVE_FORCE` rather than `I2C_SLAVE` to verify that it works?

tchilton, 2016-12-02:
I've not got that far yet. I got as far as working out what the differences were and how it worked on the stock linux tools compared to how your library works.

I've not studied your code to see how to pass the force flag in when needed, since its not required in many cases, this is why I asked raised it for your view.

I can give it a go over the next couple of days and am 99% confident it will work as its the same ioctl control that the linux tools use :-)

fivdi, 2016-12-02:
I'd suggest modifying [this line](https://github.com/fivdi/i2c-bus/blob/v1.1.2/src/setaddr.cc#L9) by replacing `I2C_SLAVE` with `I2C_SLAVE_FORCE` to see if it works first. If it does work we can decide what needs to be done.

tchilton, 2016-12-02:
I'll give it a try this evening and report back. Thanks for the response.

tchilton, 2016-12-02:
OK, as expected, it works !

The original library returns an error of "Error: EBUSY, Device or resource busy" when attempting to read two bytes from device 0x68 at offset 0x11. This is identical to running "it2get -y 1 0x68 0x11", which returns "Error: Could not set address to 0x68: Device or resource busy"

Modifying the line in your library as detailed above, recompiling results in a successful read from the RTC's temperature register, even though the kernel has the main RTC module loaded. This is identical to what I can do from the command line when adding the -f flag to i2cget 

I don't think that the I2C_SLAVE_FORCE should be the default, since this is a bit like taking the safety catch off for all I2C communication.  Whats the best way to allow this to be passed on a device by device basis, since only a couple of devices are likely to be controlled by kernel modules on any system ?

I've not looked into the inner workings of libraries in node before today so I'm not yet following how the code hangs together from the .js into the .cc files. I'm still working on this, however, the ideas I have on how this could potentially be solved are :

1. Passing a force flag on the existing i2cRead* / i2cWrite* functions, which would probably be cleaner, but presumably break existing code that uses the library. 

2. Create a new set of functions such as i2cReadForce* / i2cWriteForce* that use the other flag. Nothing else should need to be affected, hence compatibility is retained. This might create a higher maintenance overhead for the code in the future as you have similar ish functions, but I guess these already exist with the sync and async functions.

3. Some form of "safety catch" type function that can be called before reading / writing the device. The safety catch decides which ioctl is used. Given the callback nature of Node, I'm not certain how this would work if several functions were doing things at different times. 

Clearly as you know your library a lot better than me there are probably other/better/simpler/more efficient  ways to do this.  How do you want to proceed ?

fivdi, 2016-12-03:
Let me think about this for a while. Windows IoT will also have to be taken into account.

tchilton, 2016-12-03:
No worries, I'm using the workaround at the moment.

I don't know anything about the Windows model for I2C, but I assume that the ioctl's are Linux specific, so the issue may not apply to the Windows libraries.

I've done a bit of a Google around and I can't find anything matching Windows and either of the I2C_SLAVE* constants. Seems that windows does have an ioctl function, but there are no matches for this in the win folder, so I guess its either done in a different manner, or is not a problem on Windows.

Realise that you probably want to keep consistency for the library across the two platforms.

fivdi, 2016-12-04:
To enable forced access to I2C devices the open and openSync methods could be extended to support a new optional argument called options:

### open(busNumber [, options], cb)
- busNumber - the number of the I2C bus/adapter to open, 0 for /dev/i2c-0, 1 for /dev/i2c-1, ...
- options - an optional object that can be used to specify options
- cb - completion callback

### openSync(busNumber [, options])
- busNumber - the number of the I2C bus/adapter to open, 0 for /dev/i2c-0, 1 for /dev/i2c-1, ...
- options - an optional object that can be used to specify options

Initially, the optional options object could support one boolean option called forceAccess. If the forceAccess option is not specified it will have a default value of false. If forceAccess is false, access to all I2C devices using the corresponding Bus will not be forced. If forceAccess is true, access to all I2C devices using the corresponding Bus will be forced.

This would mean that each Bus instance created could be configured to either force access or not.

@munyirik as you are the Windows IoT expert, would the above modification be ok for you? My current assumption is that it's only relevant for Linux.

tchilton, 2016-12-04:
In general the approach looks good, but please note that its only the device that this applies to, not the whole bus. Consider an I2C bus with say a memory device, ADC, DAC and an RTC. Only the RTC has a kernel module, so its only that address that needs the force on it, everything else will open properly without force.

This is why I'm thinking its more appropriate for it to be an optional flag for the read and write functions for the specific devices.

fivdi, 2016-12-04:
Yes, I'm aware that the forceAccess flag would only be needed for specific devices. However, Bus objects currently have 22 methods for accessing devices at a particular address. Adding an optional forceAccess option to all 22 methods would complicate a lot of things, for example, the implementation, the documentation, understanding the documentation, ...

> Supplemental thought - were you thinking to create two objects, one with different flags to the other and all devices that need force are accessed via the force bus object ??

Correct, this is what I was thinking. To be more clear, the user of i2c-bus would could do something like this:

```js
var i2c = require('i2c-bus'),
  bus = i2c.openSync(1),
  forceAccessBus = i2c.openSync(1, {forceAccess: true});
```

tchilton, 2016-12-04:
That gets my vote !

munyirik, 2016-12-05:
@fivdi that would be ok. There isn't a way force commands with the Windows API I'm using.
But there is something called [sharing mode](https://msdn.microsoft.com/en-us/library/windows.devices.i2c.i2cconnectionsettings.sharingmode.aspx) which I was actually setting always to 'Exclusive' (see [here](https://github.com/fivdi/i2c-bus/blob/0ce5eb2a0a3e1eeac05d7570aa84c1eecc7d5008/win-i2c-bus.js#L42) - '0' as the third arg). Adding the option would be good since I can make it so that the option like {sharingmode: number} can be passed in. Is there something like this on Linux?

Here's a description of the 2 possible sharing modes ([source](https://msdn.microsoft.com/en-us/library/windows.devices.i2c.i2csharingmode.aspx)):

| Member	| Value	|Description|
|-----------|----------|------------|
Exclusive/exclusive	| 0 | Connects to the I²C bus address exclusively, so that no other connection to the I²C bus address can be made while you remain connected. This mode is the default mode.
Shared/shared | 1 | Connects to the I²C bus address in shared mode, so that other connections to the I²C bus address can be made while you remain connected. You can perform all operations on shared connections, but use such connections with care. When multiple client apps change the global state of the I²C device, race conditions can result. An example use case for using a shared connection is a sensor that obtains readings without changing the state of the device.

fivdi, 2016-12-17:
@munyirik if the documentation in the To-do list at the bottom of [this page](https://i2c.wiki.kernel.org/index.php/Main_Page) is up to date, then Linux doesn't support exclusive access to devices from user-space. Would an `exclusiveAccess` option similar to the `forceAccess` option discussed above suit your needs? Note that this `exclusiveAccess` option would then apply to all devices accessed using a particular Bus object.

munyirik, 2016-12-22:
>> Would an exclusiveAccess option similar to the forceAccess option discussed above suit your needs?

Yes

fivdi, 2017-01-06:
i2c-bus v1.2.0 which was published on npm earlier today now has a new `forceAccess` option. This option can be optionally specified when calling [open](https://github.com/fivdi/i2c-bus#openbusnumber--options-cb) or [openSync](https://github.com/fivdi/i2c-bus#opensyncbusnumber--options).

It was tested on a [C.H.I.P.](https://getchip.com/pages/chip) which has a PCF8574A I/O expander at address 0x38 on `/dev/i2c-2`. The PCF8574A is already in use by a driver.

Here's a simple test program that reads data from the PCF8574A on `/dev/i2c-2` on a C.H.I.P.

```js
var i2cBus = require('i2c-bus'),
  bus2 = i2cBus.openSync(2, {forceAccess: true});

console.log(bus2.receiveByteSync(0x38).toString(16));
```

@tchilton Please give it a try and let us know if it works.

@munyirik The Windows code wasn't modified as it has no need for the `forceAccess` option. An `exclusiveAccess` option could easily be added and wouldn't require any modification of the Linux code.

closed 2017-01-06 by fivdi

skiddaw, 2019-04-12:
This allowed another function on my bucket list to be ticked off.  On a stand-alone bat-house temperature monitor I use an LED on the i2c RTC 1Hz SQ-wave-output to indicate some sign of life.  On an HO trainset controller I use the same line to cycle a railroad crossing lamp.  Previously I was bit bashing i2c using Gordon's wiringPi. tks.