fivdi/i2c-bus #27: Module did not self-register

360disrupt, 2016-12-01:
System: `OS X 10.12.1 (Sierra)` (I'm developing on a Mac and will later use the code on a raspberry pi3)
node tested with `v6.9.0` && `v.7.2.0 (npm 4.0.3)`
`ic2 1.1.2`

Installation worked without complications. As soon as I require the module:

##Code 
```
i2c = require('i2c-bus');
```

I get the error:
```
/Users/MasterG/Projekte/1-NODEJS/pi/node_modules/bluebird/js/release/async.js:61
        fn = function () { throw arg; };
                           ^

  Error: Module did not self-register.
  
  - module.js:598 Object.Module._extensions..node
    module.js:598:18
  
  - module.js:488 Module.load
    module.js:488:32
  
  - module.js:447 tryModuleLoad
    module.js:447:12
  
  - module.js:439 Function.Module._load
    module.js:439:3
  
  - module.js:498 Module.require
    module.js:498:17
  
  - module.js:20 require
    internal/module.js:20:19
  
  - bindings.js:76 bindings
    [pi]/[bindings]/bindings.js:76:44
  
  - i2c-bus.js:4 Object.<anonymous>
    [pi]/[i2c-bus]/i2c-bus.js:4:28
  
  - module.js:571 Module._compile
    module.js:571:32
  
  - module.js:580 Object.Module._extensions..js
    module.js:580:10
  
  - module.js:488 Module.load
    module.js:488:32
  
  - module.js:447 tryModuleLoad
    module.js:447:12
  
  - module.js:439 Function.Module._load
    module.js:439:3
  
  - module.js:498 Module.require
    module.js:498:17
  
  - module.js:20 require
    internal/module.js:20:19
  
  - water-sensor.class.js:14 Object.<anonymous>
    /Users/MasterG/Projekte/1-NODEJS/pi/.tmp/serve/backend/sensor/water-sensor.
    class.js:14:9
```

360disrupt, 2016-12-02:
Currently I have this as a solution, but it does not feel like good code:
```
os = require('os')
if os.arch() == 'arm' #raspberrypi
  i2c = require('i2c-bus')
else
  console.warn "Not using I2C", os.arch()
  i2c = require('./i2c.mock.js')
```

fivdi, 2016-12-02:
I'm afraid that i2c-bus is currently only usable on Linux and Windows IoT. Support for installation on OS X was never added. The workaround that you are currently using is probably the best thing you can do for the moment. I think a proper solution would involve modification to the gyp and C++ code so that everything installs and compiles correctly on OS X. It appears to be the case that nothing compiles at the moment and this results in the `Module did not self-register` error.

360disrupt, 2016-12-02:
Ok, then I will go with this solution for now & close this.

closed 2016-12-02 by 360disrupt

julisch94, 2019-05-15:
Is there any hope, this library will be edited to work on OS X, too?