fivdi/i2c-bus #28: interrupt event - no access to INT pin like in #16

dkebler, 2016-12-08:
I have a DIO board that uses an MCP23017 and communicates via I2C.
https://www.controleverything.com/content/Digital-IO?sku=MCP23008_MCP23017_I2CDIO2 
I want to use all the pins as pullup/interrupt dry contact closures.

so  your suggestion in #16  to wire the INTA/B pin of the 23017 to a RPI interrupt capable gio pin and then use that gpio with your on/off library isn't going to work for me.

I'm new to I2C bus, on the serial bus the library I use emits an event when something shows up on the RX pin.   Then you just register a handler and your off and running.  **What happens in your library/i2c-bus.  Can ic2 slave devices even send bytes onto the bus without a master doing a read? **

So....I can infinite read say the read INTF register, then once high I can read INTCAP and figure out which pin changed.   Kinda seems silly, polling an interrupt... 

Not your problem but can't find anything that says that when 23017 fires an interrupt it not only brings up INT pin(s) high but also sends something out on the I2C bus.   If not I am sunk and must do polling.   If so how can your library emit an event for that so I can handle it.

dkebler, 2016-12-08:
Closer examination of that board and it looks like they provided pin outs for INTA/B so I can run my own wire back to to RPi and use your on/off library like in #16.

Still curious about I2c devices "pushing" bytes onto the bus without a master read.  Possible?  Library to emit events on received pushed data?

fivdi, 2016-12-08:
The interrupt pins on the MCP23017 are not broken out onto the connector on the digital IO board linked to above so you it looks the only option you have is polling. If they were broken out, they could have been used to generate interrupts on GPIO pins.

Communication on an I2C bus is always initiated by a bus master. An I2C device like the MCP23017 will never use the bus without being explicitly told to do so.

A serial port is very different to an I2C bus.

The arduino code linked to above also requires an interrupt pin from the MCP23017.

There is a pin hole on the board labelled INT which may be what's needed.

closed 2016-12-09 by fivdi