fivdi/i2c-bus #30: use case for I2cBlock

dkebler, 2016-12-16:
**are the I2cBlock read and write methods intended for use with devices like the MCP23017? ** where the cmd argument is the register byte (a chip command in essence) and then the buffer argument would contain either one or two bytes depending on whether one or both pin ports are being accessed at the same time.  http://www.futurlec.com/SFMicrochip/MCP23017.shtml

fivdi, 2016-12-16:
The use case for readI2cBlock and writeI2cBlock is accessing more than two bytes of data. To access a single byte readByte and writeByte can be used. To access two bytes readWord and writeWord can be used.

An example where readI2cBlock could be used is to access axis data on an ADXL345 accelerometer. There are six bytes of axis data at addresses 0x32 through 0x37. These six bytes of data can be read together with readI2cBlock.

dkebler, 2016-12-16:
Let me be clearer.....you seem to have methods for doing multiple byte read/write.
for example.

    bus.i2cRead(addr, length, buffer, cb)  and
   bus.readI2cBlock(addr, cmd, length, buffer, cb)

in the later (block) method there is this "cmd" argument.   I am wondering what the use of that argument is?   If is a byte/address that is sent to the device before the rest of the buffer is read (or written) then this method would be useful for communicating with the MCP23017 which wants a register address first and then the actual data to be read/written to the GPIO pins it services.

If not then I'll need to fashion my own method from those you provide.

fivdi, 2016-12-17:
I2C devices typically have a number of bytes that can be accesses. Depending on the documentation for the device, these bytes will be referred to as registers or commands. The i2c-bus module uses the term command here. Your above assumption is correct. It's the byte/address that is sent to the device before the rest of the buffer is read (or written.)

closed 2016-12-17 by dkebler