fivdi/i2c-bus #31: installed per instructions but no nodes appear?

tjohnson10010, 2017-01-30:
After appearing to install i2c-bus, I can´t find the nodes even after turning the Rpi off then on.  Any suggestions?  
I have node.js v0.10.29 and npm v2.15.11
Here is the log: 
i@raspberrypi:~ $ cd .node-red
pi@raspberrypi:~/.node-red $ npm install i2c-bus
\
> i2c-bus@1.2.0 install /home/pi/.node-red/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory '/home/pi/.node-red/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/home/pi/.node-red/node_modules/i2c-bus/build'
i2c-bus@1.2.0 node_modules/i2c-bus
├── bindings@1.2.1
└── nan@2.5.1
pi@raspberrypi:~/.node-red $

fivdi, 2017-01-30:
I'm not sure I fully understand the issue here. Do you mean Node-RED nodes? If so, the i2c-bus module doesn't implement Node-RED nodes and I wouldn't expect it to result in Node-RED nodes appearing anywhere after it is installed. I'm afraid that I'm not familiar with Node-RED and not sure that I can help here.

i2c-bus is a fairly low-level package which allows devices on an I2C bus to be accessed. If I'm not mistaken Node-RED nodes are fairly high-level packages that provide high-level access to specific components. I can imagine a Node-RED node using i2c-bus internally but don't think that i2c-bus can be used directly in Node-RED. I may however be mistaken.

tjohnson10010, 2017-01-30:
Fivdi,
  I think you nailed it.  I am working in Node-RED which has some terminology in common.  That´s how much a novice I am...
This one is solved.

closed 2017-01-30 by tjohnson10010