fivdi/i2c-bus #32: Reset i2c data address

brtthomp, 2017-02-06:
I am new to i2c communication protocol and am still getting the hang of it so sorry if this question is simple.

I have a Angstrom device where my team finally got the EEPROM working correctly.  I am able to use i2cdetect, i2cset and i2cget perfectly.  From my understanding i2cget will only read one byte from the current data address you are at and will increment the data address with every call.  i2cset will reset the data address to where ever you write to.  So for example 'i2cset -y 2 0x01 0x00 0x00 i' will reset the data address to the beginning where ic2get can be used.  This seems completely different than the implementation of i2c-bus where you are able to give the command code (which is what I am assuming is the data address?).  When I try those it doesn't work at all and seems to read from unknown data address.  If I reset the data address with i2cset before running the node process it works better but still skips the first byte for some reason.  This happens with both bus.readI2cBlock and bus.readByte.

If I use bus.i2cRead though I am able to read correctly from the start of the EEPROM data.  My main question though is how do I reset the data address to the beginning programmatically without having to call i2cset before running the node process.

fivdi, 2017-02-07:
What is the exact type of the EEPROM being used, preferably with a link to the datasheet for that EEPROM?

brtthomp, 2017-02-07:
I think I posted a little too early.  The issue was with the fact that the EEPROM I am using has 16bit addressing instead of 8bit which I forgot to account for.

closed 2017-02-07 by brtthomp