fivdi/i2c-bus #33: [FR]switch from node-gyp to node-pre-gyp

farfromrefug, 2017-03-07:
That would be awesome if you could switch from node-gyp to [node-pre-gyp](https://github.com/mapbox/node-pre-gyp).
that way we could have pre build release and make ```npm install``` so much faster!
If you need an example [node-usb](https://github.com/tessel/node-usb) uses it

Thanks

fivdi, 2017-03-07:
Although there are exceptions, i2c-bus is typically installed on single board computes with ARM processors. To the best of my knowledge there are no node modules that use node-pre-gyp with binaries for ARM architectures. I've only ever seen it being used for binaries for ia32 and x64 Intel architectures. The binaries available for some popular modules are, for example, those for [leveldown](https://github.com/Level/leveldown/releases/tag/v1.6.0), [node-serialport](https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/tag/4.0.7) and [usb](https://github.com/tessel/node-usb/releases/tag/1.2.0). As can be seen, binaries are available for the ia32 and x64 architectures but not for any of the ARM archictures. It can also be seen that there are thirty different binaries for a single version of serialport. Note that there are quite a number of ARM architectures, there's ARMv6, ARMv7-A, ARMv8 (32 & 64 bit), ...

To be honest, given the large effort that would be required to implement this feature, it's not something that I'm prepared to work on. Also, given that I haven't seen node-pre-gyp being used for binaries for any of the ARM architectures, I'm even less inclined to work on the topic.

In additon, it only takes a little more that 16 seconds to install i2c-bus on a Raspberry Pi 3. This may not be blazing fast, but it's not bad.
```
pi@raspberrypi:~/node-v7.7.1 $ time npm install i2c-bus

> i2c-bus@1.2.1 install /home/pi/node-v7.7.1/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory '/home/pi/node-v7.7.1/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/home/pi/node-v7.7.1/node_modules/i2c-bus/build'
/home/pi/node-v7.7.1
└─┬ i2c-bus@1.2.1 
  ├── bindings@1.2.1 
  └── nan@2.5.1 

npm WARN enoent ENOENT: no such file or directory, open '/home/pi/node-v7.7.1/package.json'
npm WARN node-v7.7.1 No description
npm WARN node-v7.7.1 No repository field.
npm WARN node-v7.7.1 No README data
npm WARN node-v7.7.1 No license field.

real	0m16.367s
user	0m16.020s
sys	0m0.810s
pi@raspberrypi:~/node-v7.7.1 $ 
```

closed 2017-03-07 by fivdi

farfromrefug, 2017-03-08:
[Same thing](https://github.com/fivdi/pigpio/issues/17)

BTW what pi did you tested on? It s really slower on a Pi Zero than on a Pi2

fivdi, 2017-03-11:
As mentioned above, the test was performed on a Raspberry Pi 3.