fivdi/i2c-bus #35: Use in Node-Red

scargill, 2017-07-30:
Hi

In order to test your I2c code (having seen onoff in action thanks to node-red-contrib-opi-gpio)

I thought I'd run a test.... in node red. So in the settings file I did the usual - as you can't REQUIRE in node-red itself....

At the start of settings I added i2c in the same way I add filesystem

var fs = require("fs");
var i2c = require("i2c-bus");

And further down I make it available in the same way I do with filesystem...

    functionGlobalContext: {
        os:require('os'),
        moment:require('moment'), 
        fs:require('fs'), 
        i2c:require("i2c-bus"),
        mySettings:mySettings 
    },

And all of that is fine - Node-Red starts up wonderfully..

However in taking your example.... and putting it in a function in Node-red

 var i2c=global.get("i2c");
  i2c1 = i2c.openSync(1);

var DS1621_ADDR = 0x48,
  CMD_ACCESS_CONFIG = 0xac,
  CMD_READ_TEMP = 0xaa,
  CMD_START_CONVERT = 0xee;

function toCelsius(rawTemp) {
  var halfDegrees = ((rawTemp & 0xff) << 1) + (rawTemp >> 15);

  if ((halfDegrees & 0x100) === 0) {
    return halfDegrees / 2; // Temp +ve
  }

 
etc as per your example,,, I get the dreaded permissions issue.

30 Jul 14:01:46 - [info] Started modified nodes
30 Jul 14:01:52 - [error] [function:ded5b043.8072a] Error: EACCES: permission denied, open '/dev/i2c-1'

Looks to me like Node-Red is understanding this just fine - but is not allowed to access it.   In the onoff situation - a 99 file was required to ensure access for Node-Red..

Here's what was needed for GPIO. Any idea of an equivalent that might be needed to enable pi access to the I2c??? 

User setup
To meet these requirements for user pi, run this commands as root: Check if user pi belongs to gpio:

$ groups pi
Otherwise, run this:

$ addgroup gpio
$ usermod -a -G gpio pi
If pi is not allowed to use GPIO now, additional steps are required:

Kernel Setup
Create a new file:

$ touch /etc/udev/rules.d/99-com.rules
copy this text to the file:

KERNEL=="gpio*", RUN="/bin/sh -c 'chgrp -R gpio /sys/%p /sys/class/gpio && chmod -R g+w /sys/%p /sys/class/gpio'" 
and do a reboot.

scargill, 2017-07-30:
Ignore me - adding pi to group i2c seems to have done the job - I now get a timeout as the device isn't actually connected. Good start.

fivdi, 2017-07-30:
Correct, in order to access I2C devices on the Pi the user needs to be added to group i2c.

There doesn't appear to be a TODO for i2c-bus here so I'll close the issue.

closed 2017-07-30 by fivdi

nielsnl68, 2017-09-26:
I created my second version of my node-red-contrib-i2c nodes around your library. And i really like it.

fivdi, 2017-09-26:
> I created my second version of my node-red-contrib-i2c nodes around your library. And i really like it.

@nielsnl68 Thanks :)