fivdi/i2c-bus #36: i'm getting lots of "Error: , Remote I/O error"

nielsnl68, 2017-09-26:
Hello,

lately i created a set of node-red nodes with your library as basis. And everything works very well most of the time. Except i'm getting "Error: , Remote I/O error" error sometimes.
I use node-red on a Raspberry Pi3 connected to a couple of arduino's. 

Not sure where the issue is coming from.

fivdi, 2017-09-26:
All variants of the Raspberry Pi have an [I2C hardware bug](http://www.advamation.com/knowhow/raspberrypi/rpi-i2c-bug.html) which prevents I2C clock stretching from functioning correctly. A symptom of this bug is "remote I/O errors". To avoid this issue do not use I2C devices which stretch the I2C clock with a Raspberry Pi.

Arduino UNOs are not particularly fast and it's relatively easy to implement code that is not fast enough to prevent an Arduino UNO from stretching the I2C clock.

I would imagine that you are seeing this issue here.

The default baudrate on the raspberry Pi is 100000. It can be lowered to 10000 by adding the following line to `/boot/config.txt` and rebooting the Pi.
```
dtparam=i2c_baudrate=10000
```
Give this a try to see if it helps.

If it does help you could increase the baudrate bit by bit to determine what the max baudrate is that functions correctly.

An alternative would be to catch the error and retry.

fivdi, 2017-09-29:
There's no todo for i2c-bus here so I'll go ahead and close the issue.

closed 2017-09-29 by fivdi

emilbayes, 2017-10-05:
This issue helped me too. Thanks!

fivdi, 2018-04-28:
If you are having issues communicating with an I2C device connected to a Raspberry Pi and the issues are related to the hardware I2C clock stretching bug consider using software I2C rather than hardware I2C. See [Configuring Software I2C on the Raspberry Pi](https://github.com/fivdi/i2c-bus/blob/master/doc/raspberry-pi-software-i2c.md) for further details.

msi-matthew, 2018-09-20:
@fivdi Do you have an example of the a catch/retry?

fivdi, 2018-09-20:
There's an example using try/catch in [i2cscan.js](https://github.com/fivdi/i2c-bus/blob/master/example/i2cscan.js). The example doesn't retry but shouldn't be too difficult to figure out.