fivdi/i2c-bus #37: Installing the i2c-bus

nielsnl68, 2017-10-08:
Hi, i try to install this lib on my windows system but i get the following error.
Did you try to install this on a windows system?

`C:\...\.node-red\node_modules\i2c-bus>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "" rebuild )
Warning: unrecognized setting VCCLCompilerTool/CompileAsWinRT
Warning: unrecognized setting VCCLCompilerTool/CompileAsWinRT
De projecten in deze oplossingen worden één voor één opgebouwd. Als u parallelle bouwbewerkingen wilt inschakelen, gebruikt u de schakeloptie /m.
MSBUILD : error MSB3428: kan het Visual C++-onderdeel VCBuild.exe niet laden. U
 kunt dit oplossen door 1) de .NET Framework 2.0 SDK te installeren, 2) Microso
ft Visual Studio 2005 te installeren of 3) de locatie van het onderdeel toe te
voegen aan het systeempad, als het onderdeel elders is geïnstalleerd.  [C:\...\.node-red\node_modules\i2c-bus\bu..ild\binding.sln]
gyp ERR! build error
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:219:12)
gyp ERR! System Windows_NT 10.0.15063
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\niels\.node-red\node_modules\i2c-bus
gyp ERR! node -v v6.11.4
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
npm WARN node-red-project@0.0.1 No repository field.
npm WARN node-red-project@0.0.1 No license field.
npm ERR! Windows_NT 10.0.15063
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "i2c-bus"
npm ERR! node v6.11.4
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! i2c-bus@1.2.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the i2c-bus@1.2.2 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the i2c-bus package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs i2c-bus
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls i2c-bus
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\....\npm-debug.log`

fivdi, 2017-10-13:
i2c-bus doesn't support windows. It supports Linux and Windows IoT.

munyirik, 2017-10-13:
This should be working. Since the Windows IoT Core OS doesn't support msbuild, it's required to do the npm install on a Windows 10 machine and then copy over to a Windows IoT device. 
Also since Windows IoT Core is based on Windows 10, i2c-bus should be able to run on Windows 10 desktop but I never tested this.
I'm currently swamped but I'll try and get some time in the next couple of days to investigate.

munyirik, 2017-10-13:
cc @paulmon

munyirik, 2017-10-13:
I took a quick look to get an idea of what's wrong...

1. CompileAsWinRT support hasn't reached node-v8 yet. It's needed to build the Windows i2c code correctly. node-v8 runs on x86 and x64 versions of Windows IoT Core but not ARM. I'd suggest switching to [node-chakracore](https://github.com/nodejs/node-chakracore) and that should will get you past the first error.
2. There's another problem where binding.gyp needs to be updated. In summary the /ZW flag isn't getting set and the path to platform.winmd isn't getting set (this [blog](http://blogs.microsoft.co.il/pavely/2012/09/29/using-ccx-in-desktop-apps/) has some details of what I just mentioned)
3. Unfortunately even after the issues above are fixed there's another problem. NAN references v8:JSON which is missing from the node-chakracore version of v8.h. I've opened [another issue](https://github.com/nodejs/node-chakracore/issues/416) for this problem.

munyirik, 2017-10-14:
Not sure when I can have a fix for this but to get unblocked and build i2c.node independently, here's what you can do - sorry for the inconvenience :-(

1. Install node-chakracore 8.1.2-pre2.
2. Clone i2c-bus repo.
3. cd to the i2c-bus directory
4. Install node-gyp with 'npm install -g node-gyp'
5. Install NAN with 'npm install nan'
6. Run 'node-gyp configure' (this will create build\binding.sln)
7. Open the solution file and make the following changes...
8. In Properties (Linker/General/Additional Library Directories), add C:\Users\foo\AppData\Local\nvs\chakracore\8.1.2-pre2\x64\node-v8.1.2-nightly201706274be96a0e53-win-x64\sdk\Release (this should contain chakracore.lib)
9. In Properties (C/C++/General/Additional #using Directories) add C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.11.25503\lib\x86\store\references. This might be different depending on the version of VS you have. This location points has platform.winmd.
10. In Properties (C/C++/General/Consume Windows Runtime Extension) select Yes (/ZW)

After these steps, the solution should build i2c.node successfully.

fivdi, 2017-10-14:
@munyirik Thanks for looking into this.

fivdi, 2018-03-09:
@munyirik any new on this one?

munyirik, 2018-03-20:
Hey @fivdi sorry no update yet. Been swamped with other work.

fivdi, 2018-04-18:
I'm going to close this as a `"wontfix"`. Unfortunately nobody has the bandwidth to fix it at the moment. Sorry.

closed 2018-04-18 by fivdi