fivdi/i2c-bus #39: {"errno":55,"code":"EIO","syscall":"write"}

MaBecker, 2017-12-15:
beaglebone black 3.8.13-bone84
node v0.10.38, 
i2c-bus version 1.2.1

this is the used sample:

```
var i2c  = require('i2c-bus');
var i2cbusInst = i2c.openSync(self.bus);
i2cbusInst.sendByteSync(self.id, val);
i2cbusInst.closeSync();
```

Any hits how to avoid this error`

fivdi, 2017-12-15:
There are three calls to i2c-bus methods in the above code. Which of the calls results in the error? It would be better to post a complete error message that includes a stack trace rather than just `{"errno":55,"code":"EIO","syscall":"write"}`. This would make it easier to see where the error occurred.

The errno in the message is 55 and the code is EIO. This is a strange. For code EIO I would have expected to see errno 5. errno 55 corresponds to code ENOANO.
See `/usr/include/asm-generic/errno.h`
and `/usr/include/asm-generic/errno-base.h`

The sendByteSync method implements the SMBus send byte protocol and is used to send a single byte to an I2C device. Does the I2C device that the BeagleBone Black is communicating with support the SMBus send byte pprotocol? Please check the documentation for the I2C device to see if this is the case. There is a short description of the SMBus send byte protocol on page 4 of [this document](http://www.atmel.com/Images/doc2583.pdf).

MaBecker, 2017-12-16:
Yep, errno and code do not match, this is really weird.

Will change to i2cWriteSync() and i2cReadSync() and continue testing

fivdi, 2017-12-16:
> Used i2c device can do I2C and SMBus.

I'm afraid that this isn't a precise enough answer.

There are many different SMBus protocols. For example here are several different protocols related to writing data:
 * Quick command
 * Send byte
 * Write byte/word
 * Block write

These protocols all serve different purposes. For example, the send byte protocol is very different to the write byte protocol.

When the datasheet for an I2C device says it supports SMBus this doesn't mean it supports all SMBus protocols. All the different SMBus protocols are defined in the [System Management Bus (SMBus) Specification](http://smbus.org/specs/smbus20.pdf).

Please post a link to the datasheet for the I2C device that is being used and indicate where support for the SMBus send byte protocol is mentioned in the datasheet.

Also, as mentioned above a complete error message including a stack trace is needed to determine where the error actually occurred.

fivdi, 2017-12-16:
Is this issue occurring with the OLED display mentioned in #38 or is this issue for a different device?

MaBecker, 2017-12-16:
have to upgrade os an node - thanks for your support

closed 2017-12-16 by MaBecker

fivdi, 2017-12-16:
> have to upgrade os an node - thanks for your support

No problem.

I took a closer look at how Node.js handles errors from the operating system. By the looks of things some of the work is done by uv rather than Node.js. As can be seen in the source code of Node.js v0.10.38 [here](https://github.com/nodejs/node-v0.x-archive/blob/v0.10.38/deps/uv/include/uv.h#L128) errno 55 is a uv error number rather than a Linux error number. It corresponds to Linux error number 5 which is EIO. This explains the 55.