fivdi/i2c-bus #41: Communication problem with SHT25 sensor

poledeveloper, 2018-01-04:
Hi Brian!
I’m trying to implement your I2C serial bus access for Node.js to communicate with SHT25 sensor on Raspberry PI 3
(https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/2_Humidity_Sensors/Sensirion_Humidity_Sensors_SHT25_Datasheet.pdf)
but I don't understand where i'm doing wrong!

This is my edit version of your original code! The error is already at the call of the writeByteSync method.
https://drive.google.com/file/d/190aax_Fu8SSw-HPbREk0g3wuDNLYbgwy

This is the error generated:

/home/pi/Documents/node_js_projects/i2c/i2c-bus/node_modules/i2c-bus/i2c-bus.js:257
  i2c.writeByteSync(peripheralSync(this, addr), cmd, byte);
      ^
Error: , Remote I/O error
    at Bus.writeByteSync (/home/pi/Documents/node_js_projects/i2c/i2c-bus/node_modules/i2c-bus/i2c-bus.js:257:7)
    at /home/pi/Documents/node_js_projects/i2c/i2c-bus/i2c-bus.js:24:8
    at Object.<anonymous> (/home/pi/Documents/node_js_projects/i2c/i2c-bus/i2c-bus.js:26:2)
    at Module._compile (module.js:635:30)
    at Object.Module._extensions..js (module.js:646:10)
    at Module.load (module.js:554:32)
    at tryModuleLoad (module.js:497:12)
    at Function.Module._load (module.js:489:3)
    at Function.Module.runMain (module.js:676:10)
    at startup (bootstrap_node.js:187:16)

I don't know if this will help or not, this is the python code that works perfectly.
https://github.com/ControlEverythingCommunity/SHT25/blob/master/Python/SHT25.py

I'm hoping you can help me!
Thanks in advance!

fivdi, 2018-01-04:
Hi Federico,

The `Remote I/O error` indicates that program is having issues communicating with the sensor.

This is what the program looks like at the moment:

```js
var i2c = require('i2c-bus'),
  i2c1 = i2c.openSync(1);

var SHT25_ADDR = 0x40,
  CMD_ACCESS_CONFIG = 0xF3;

function toCelsius(rawTemp) {
  var halfDegrees = ((rawTemp & 0xff) << 1) + (rawTemp >> 15);

  if ((halfDegrees & 0x100) === 0) {
    return halfDegrees / 2; // Temp +ve
  }

  return -((~halfDegrees & 0xff) / 2); // Temp -ve
}

(function () {
  var rawTemp;

  console.log('start writeByteSync:');

  // Enter one shot mode (this is a non volatile setting)
  i2c1.writeByteSync(SHT25_ADDR, CMD_ACCESS_CONFIG, 0x01);

}());
```

It looks like the program is calling `i2c1.writeByteSync(SHT25_ADDR, CMD_ACCESS_CONFIG, 0x01)` to trigger a temperature measurement.

The `writeByteSync` method implements the SMBus write byte protocol which looks like this:
```
S Addr Wr [A] Comm [A] Data [A] P
```
However, the datasheet for the SHT25 indicates that the SMBus send byte command is used to trigger a temperature measurement. The SMBus send byte command looks like this:
```
S Addr Wr [A] Data [A] P
```
Please replace this:
```js
  i2c1.writeByteSync(SHT25_ADDR, CMD_ACCESS_CONFIG, 0x01);
```
with this:
```js
  i2c1.sendByteSync(SHT25_ADDR, CMD_ACCESS_CONFIG);
```
to see if it works. I think this will trigger the temperature measurement and that the `Remote I/O error` should no longer occur.

Please also note that the code in `toCelsius` appears to have been copied from the i2c-bus README and will only function for a DS1621 temperature sensor. For a SHT25 the algorithm is different.

The python program calls `read_byte` twice to read the temperature. The equivalent method in i2c-bus is `receiveByteSync`. It should also be possible to use `i2cRead` to read both bytes at the same time.

poledeveloper, 2018-01-05:
You're exactly right, Brian.
With the following code it works fine:

'''
var i2c = require('i2c-bus'), i2c1 = i2c.openSync(1);

var SHT25_ADDR = 0x40, CMD_GET_TEMP = 0xF3;

function sleep(ms)
{
  return new Promise(resolve => setTimeout(resolve, ms));
}

async function readAfterSleep(ms)
{

	console.log('Taking a break...');
	await sleep(ms);
	console.log(ms + ' millisecond later');

	// SHT25 address, 0x40(64)
	// Read data back, 2 bytes
	// Temp MSB, Temp LSB
	console.log('start receiveByteSync:');
	var MSB = i2c1.receiveByteSync(SHT25_ADDR);
	var LSB = i2c1.receiveByteSync(SHT25_ADDR);
	console.log('receiveByteSync MSB: ' + MSB);
	console.log('receiveByteSync LSB: ' + LSB);
	console.log('celsiusTemperature from receiveByteSync: ' + toCelsius(MSB, LSB));

	i2c1.closeSync();

}

function toCelsius(data0, data1)
{
	var temp = data0 * 256 + data1;
	var celsiusTemperature= -46.85 + ((temp * 175.72) / 65536.0);
	return celsiusTemperature;
}

(function ()
{
	var rawTemp;
  	console.log('start writeByteSync:');
	i2c1.sendByteSync(SHT25_ADDR, CMD_GET_TEMP);
	readAfterSleep(500);
}());
'''

Thanks for the quick reply!
Federico

fivdi, 2018-01-05:
>With the following code it works fine

Excellent.
>without sleep function, receiveByteSync don't works

This is to be expected and is mentioned in the SHT25 data sheet.

It looks like all is ok now so I'll go ahead and close the issue.

closed 2018-01-05 by fivdi