fivdi/i2c-bus #42: not able to catch bad arguments error

dkebler, 2018-02-09:
Tried several ways....but not able to catch the bus call error (when I deliberately send a bad or missing argument) before it terminates the node process. 
Note that the    `console.log('===>',err)` is never executed meaning the process terminated before the callback was called.

Is this because the bus call is really a C call and the node process can't control the C call terminating the process?

The process as it is a socket server and need to guard against sending one packet with a bad set of arguments crashing the server. 

I even restored back to error first callback.
```
  write: function (args) {
    return new Promise((resolve) => {
      this.i2cbus.writeByte.bind(this.i2cbus)(args.address, args.cmd, args.byte,
        (err,res) => {
          console.log('===>',err)
          if (err) resolve({error:err.msg})
          resolve(res ||'success')
        })
    })
  }
```

```
  i2c.setAddrAsync(device, addr, bus._forceAccess, function (err) {
          ^

Error: EINVAL, incorrect arguments passed to setAddr(int fd, int addr, bool forceAccess, function cb)
    at /opt/uci/uci-i2c-bus/node_modules/i2c-bus/i2c-bus.js:78:11
    at FSReqWrap.oncomplete (fs.js:149:20)

```

dkebler, 2018-02-10:
even tried wrapping the writeByte call in a try catch block and that did not work to catch the error.

fivdi, 2018-02-10:
@dkebler Thank you for filing this issue. It's a bug and there isn't really anything that you can do in your code to handle or catch the error.

The C++ function [SetAddrAsync](https://github.com/fivdi/i2c-bus/blob/v1.2.5/src/setaddr.cc#L41-L53) which is called from JavaScript [here](https://github.com/fivdi/i2c-bus/blob/v1.2.5/i2c-bus.js#L78) throws an error if any of it's parameters are invalid. Unfortunately the JavaScript code doesn't catch the error so it gets swallowed.

I'll take look at fixing it asap.

dkebler, 2018-02-10:
Thx,

It's harsh but for now I resorted to launching with pm2 which will restart on crash (which nodemon will not).  Crashing dumps my clients but I needed to implement a ping/reconnect anyway for client.

fivdi, 2018-02-10:
if [this](https://github.com/fivdi/i2c-bus/blob/v1.2.5/i2c-bus.js#L78-L84) block of code is changed from:
```js
      i2c.setAddrAsync(device, addr, bus._forceAccess, function (err) {
        if (err) {
          return cb(err);
        }

        cb(null, device);
      });
```
to:
```js
      try {
        i2c.setAddrAsync(device, addr, bus._forceAccess, function (err) {
          if (err) {
            return cb(err);
          }

          cb(null, device);
        });
      } catch (err) {
        cb(err);
      }
```
it should work.

Note that there are several other C++ functions that are similar and where a similar modification will be needed.

fivdi, 2018-02-10:
> Note that there are several other C++ functions that are similar and where a similar modification will be needed.

I'll fix those also.

fivdi, 2018-02-10:
@dkebler The modification with try/catch that I suggested making above isn't the solution. This solution will only work if the correct number of parameters are passed to functions and the last parameter is the required callback. If too few or to many parameters are passed or if the required callback isn't in the correct position it will not work.

dkebler, 2018-02-10:
I had already written some verification code that verifies all the required arguments in the the json packet received are present before making a particular call.  It needs debugging.  Guess I better get on that.

On your side it's kinda hard to vet the arguments but at least you could check for the correct argument count for each function and that the last is a function (in async cases).  It would be hard to distinguish between an address, cmd, and byte as they are all the same type and range so no way really to check which one might be missing when argument count is off.   The ones that take buffer could be checked for that.  Even this little would be very helpful  I can do what I can on my side but having whatever verification is possible is good considering there is no way to recover from this.

If you choose to do nothing then I'd at least but a clear warning in the readme about having to do your own argument verification before a call to avoid unrecoverable errors.

----
So then what would the try catch block above catch? a bad I2c address?  i.e no device for that address.  What exactly would that catch?

dkebler, 2018-02-10:
for buffer and function check lodash has standalone
https://www.npmjs.com/package/lodash.isfunction
https://www.npmjs.com/package/lodash.isbuffer

fivdi, 2018-02-10:
Thanks for the links.

fivdi, 2018-02-11:
@dkebler i2c-bus@2.0.0 has been published on npm. It checks parameters at the API surface and the issues that you were having should be resolved. i2c-bus should no longer throw errors due to invalid parameters that can't be handled. It would be great if you could give it a try and report your findings.

fivdi, 2018-02-14:
Fixed with i2c-bus@2.0.0

closed 2018-02-14 by fivdi

fivdi, 2018-02-24:
It's probably best to explain this with an example.

The first point is that it's necessary to upgrade to i2c-bus v2.0.0 to get the bug fix so `package.json` will most likely need to be modified.

Lets assume that we want to communicate with an MCP9808 temperature sensor at address 0x18. An MCP9808 has a one byte resolution register (command) at 0x08. The default content of this register is 3. Here's a test program that uses promises to read register 0x08.

```js
const i2c = require('i2c-bus');
const i2c1 = i2c.openSync(1);

const readByte = (i2cBus, i2cAddress, register) => {
  return new Promise((resolve, reject) => {
    i2cBus.readByte(i2cAddress, register, (err, byte) => {
      if (err) {
        reject(err);
      } else {
        resolve(byte);
      }
    });
  });
}

readByte(i2c1, 0x18, 0x08).then((byte) => {
  console.log(byte);
}).catch((err) => {
  console.log('Oops');
  console.log(err.stack);
});
```

Here `"readByte(i2c1, 0x18, 0x08).then..."` is called to read register 0x08 and everything functions correctly. The output of the program is:
```
3
```

If `"readByte(i2c1, 0x18, 0x08).then..."` is changed to `"readByte(i2c1, 0x18, "hello").then..."` the output of the program with i2c-bus v1.2.5 will be:
```
/home/pi/i2c-bus-1.2.5/node_modules/i2c-bus/i2c-bus.js:165
    i2c.readByteAsync(device, cmd, cb);
        ^

Error: EINVAL, incorrect arguments passed to readByte(int fd, int cmd, function cb)
    at Error (native)
    at Bus.<anonymous> (/home/pi/i2c-bus-1.2.5/node_modules/i2c-bus/i2c-bus.js:165:9)
    at /home/pi/i2c-bus-1.2.5/node_modules/i2c-bus/i2c-bus.js:83:9
```

The exception handling does not function correctly with i2c-bus v1.2.5. It's not possible to catch this exception. Note how the text "Oops" is not logged to the console. Using try/catch will not help either.

If `"readByte(i2c1, 0x18, 0x08).then..."` is changed to `"readByte(i2c1, 0x18, "hello").then..."` the output of the program with i2c-bus v2.0.0 will be:
```
Oops
Error: Invalid I2C command hello
    at checkCommand (/home/pi/i2c-bus/node_modules/i2c-bus/i2c-bus.js:25:11)
    at Bus.readByte (/home/pi/i2c-bus/node_modules/i2c-bus/i2c-bus.js:236:3)
    at Promise (/home/pi/i2c-bus/t.js:8:12)
    at readByte (/home/pi/i2c-bus/t.js:7:10)
    at Object.<anonymous> (/home/pi/i2c-bus/t.js:18:1)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
```

The exception handling function correctly with i2c-bus v2.0.0. It's possible to catch the exception. Note how the text "Oops" is logged to the console.

Exception handling was modified for all i2c-bus methods and they should all function correctly and play well with promises now.

i2c-bus does not check that the number of arguments passed to methods is exactly equal to the required number of arguments. It checks to see if the required arguments are correct. If too many arguments are passed the additional arguments that are not required are ignored.