fivdi/i2c-bus #44: 3.0.0 wont build

farfromrefug, 2018-03-02:
I can't build 3.0.0
I get those errors:
```
In file included from ../src/i2c.cc:96:0:
../src/./readbyte.cc: In member function 'virtual void ReadByteWorker::HandleOKCallback()':
../src/./readbyte.cc:34:29: error: 'async_resource' was not declared in this scope
     callback->Call(2, argv, async_resource);
```
I don't see where ```async_resource``` is defined

fivdi, 2018-03-02:
`async_resource` is defined in [here](https://github.com/nodejs/nan/blob/v2.9.2/nan.h#L1762) in nan.h.

Can you post the command that was used to install `i2c-bus` please?

fivdi, 2018-03-02:
In addition to posting the command used to install `i2c-bus` please post the complete output of that command.

farfromrefug, 2018-03-02:
Thanks now i found why! I use yarn which did not update the ```nan``` version. I was still in ```2.7.0```
That will fix the build

closed 2018-03-02 by farfromrefug