fivdi/i2c-bus #46: ramdom bitfilp in data read from bno055 i2c sensor

khalifima, 2018-04-17:
Hi
I am using raspberry pi zero w (bcm2835) to which i am connecting a bno055 sensor, using readByteSync() to read from the sensor,  I experience an odd behavior: somehow in a random way some values have their most significant bit flipped from zero to 1 although the data is being correctly sent from the sensor when I check it using the data analyzer. i am using node 8.x in Raspbian stretch lite 03-2018.

fivdi, 2018-04-17:
Please post code that can be used to reproduce the problem.

khalifima, 2018-04-17:
Update: I have created a new i2c-gpio device using the provided overlay from raspbian. The created device was under /dev/i2c-3 and I used it and it seems to work without problems.

fivdi, 2018-04-17:
All variants of the Raspberry Pi (today) have an [I2C hardware bug](http://www.advamation.com/knowhow/raspberrypi/rpi-i2c-bug.html) which prevents I2C clock stretching from functioning correctly. Unfortunately the BNO055 relies on I2C clock stretching. Someone describes their experiences with I2C and the BNO055 on a Raspberry Pi [here](http://projects.webvoss.de/2017/03/19/motion-sensor-poc-bno055-and-raspberry-pi-subtleties/).

Thank you very much explaining how you were able to resolve the issue using an i2c-gpio device. This is very useful information. An i2c-gpio device uses software rather than hardware to implement I2C and it looks like it handles I2C clock stretching correctly.

fivdi, 2018-04-28:
The [Installation section of the README](https://github.com/fivdi/i2c-bus#installation) now links to the new document [Configuring Software I2C on the Raspberry Pi](https://github.com/fivdi/i2c-bus/blob/master/doc/raspberry-pi-software-i2c.md).

closed 2018-04-28 by fivdi