fivdi/i2c-bus #47: Device ID (re-send start)

johntalton, 2018-04-17:
Is it possible with this lib to achieve commands that require re-send of start.   Most notably the Device ID common function (3.1.17 of https://www.nxp.com/docs/en/user-guide/UM10204.pdf)

 thanks

fivdi, 2018-04-18:
While this library does support repeated start for the specific use cases listed below it doesn't support Device ID use case.
```js
    bus.readByte(addr, cmd, cb)
    bus.readByteSync(addr, cmd)
    bus.readWord(addr, cmd, cb)
    bus.readWordSync(addr, cmd)
    bus.readI2cBlock(addr, cmd, length, buffer, cb)
    bus.readI2cBlockSync(addr, cmd, length, buffer)
```

johntalton, 2018-04-18:
moving forward is the intent to introduce transaction api? Wire.h exposes one on top of twi (not my fav idea)

alt. could introduce api for devId and Gcr etc directly

it looks like the defined for RDWR are stubbed in to access, no?

thoughts? thanks

fivdi, 2018-04-18:
I haven't thought about the topic a much but here are the initial thoughts.

> moving forward is the intent to introduce transaction api?

Probably not.

> alt. could introduce api for devId and Gcr etc directly

This is more likely as it would be consistent with the current API.

> it looks like the defined for RDWR are stubbed in to access, no?

I don't fully understand this question. Is part of the question is missing? If the question is whether combined read/write transaction without stop in between using `ioctl(file, I2C_RDWR, struct i2c_rdwr_ioctl_data *msgset)` is already available in i2c-bus then the answer is no. Currently `ioctl(file, I2C_SMBUS, struct i2c_smbus_ioctl_data *args)` is used for most operations.

johntalton, 2018-04-19:
open PR #48

fivdi, 2018-04-21:
@johntalton Thank you very much for adding this feature. The PR has been merged is available with i2c-bus@3.2.0 on npm.

closed 2018-04-21 by fivdi