fivdi/i2c-bus #50: Native module doesn't compile against Node.js v10.2.0

tvdstaaij, 2018-05-24:
v10.1 works fine but both my build server and local machine can't install i2c-bus with v10.2. Log:

```
$ node --version
v10.2.0
$ npm --version
5.6.0
$ npm install

> i2c-bus@3.2.0 install /home/user/git/node-i2c-adt7420/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory '/home/user/git/node-i2c-adt7420/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
In file included from ../src/i2c.cc:1:0:
/home/user/.node-gyp/10.2.0/include/node/node.h:53:10: fatal error: core.h: No such file or directory
 #include "core.h"  // NOLINT(build/include_order)
          ^~~~~~~~
compilation terminated.
make: *** [i2c.target.mk:97: Release/obj.target/i2c/src/i2c.o] Error 1
make: Leaving directory '/home/user/git/node-i2c-adt7420/node_modules/i2c-bus/build'
```

tvdstaaij, 2018-05-24:
Looks like this is a nodejs / node-gyp bug: https://github.com/nodejs/node/issues/20921

fivdi, 2018-05-24:
@tvdstaaij Thanks for the report.

Lets leave this issue open until Node.js v10.2.1 is released with a [fix](https://github.com/nodejs/node/issues/20921#issuecomment-391578530).

fivdi, 2018-05-25:
Node.js v10.2.1 has been release and is available for download at https://nodejs.org/dist/v10.2.1/.
It should be possible to install i2c-bus with this release without errors.

closed 2018-05-25 by fivdi