fivdi/i2c-bus #52: Native module does not compile against node 10.4.1

rektide, 2018-06-19:
Relevant logs (i think?) from trying to npm install:

```
> node-gyp rebuild

make: Entering directory '/home/rektide/src/hearty-wonder-light/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
In file included from ../../nan/nan.h:190:0,
                 from ../src/i2c.cc:3:
../../nan/nan_maybe_43_inl.h: In function ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’:
../../nan/nan_maybe_43_inl.h:88:15: error: ‘class v8::Object’ has no member named ‘ForceSet’
   return obj->ForceSet(GetCurrentContext(), key, value, attribs);
               ^~~~~~~~
```

rektide, 2018-06-19:
I tried 10.2 (since it was reported as being fixed in #50) but that didn't work. But I am getting a different error than what was seen there.

I had been on 8.9.4 and that was working. I just switched back and it again works.

This is debian linux on amd64.

fivdi, 2018-06-19:
i2c-bus installs successfully for me with Node.js v10.4.1:
```
pi@raspberrypi:~/i2c-bus $ node -v
v10.4.1
pi@raspberrypi:~/i2c-bus $ npm i i2c-bus

> i2c-bus@3.2.0 install /home/pi/i2c-bus/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory '/home/pi/i2c-bus/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/home/pi/i2c-bus/node_modules/i2c-bus/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/i2c-bus/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/i2c-bus/package.json'
npm WARN i2c-bus No description
npm WARN i2c-bus No repository field.
npm WARN i2c-bus No README data
npm WARN i2c-bus No license field.

+ i2c-bus@3.2.0
added 3 packages from 9 contributors and audited 3 packages in 22.207s
found 0 vulnerabilities

pi@raspberrypi:~/i2c-bus $ 
```

The issue is related to the i2c module, not the i2c-bus module. See [here](https://github.com/kelly/node-i2c/issues/90).

Why does the [hearty-wonder-light module use both i2c and i2c-bus](https://github.com/rektide/hearty-wonder-light/blob/master/package.json#L26-L27)? I would have thought either one or the other would be used but not both.

rektide, 2018-06-19:
oh whoops sorry, i forgot to get rid of the i2c dep. thankssssssss sorry. indeed, i only need i2c-bus.

closed 2018-06-19 by rektide