fivdi/i2c-bus #53: Error: Cannot find module 'i2c-bus' after running our node.js

vkjuju, 2018-07-24:
We hit a weird error(Error: Cannot find module 'i2c-bus'
)  even we have done" npm install i2c-bus",  we have also checked: 
$npm list|grep i2c-bus
├─┬ i2c-bus@3.2.0
│ ├─┬ i2c-bus@1.2.5
│ ├── i2c-bus@3.2.0 deduped
any advice would be appreciated.

Error: Cannot find module 'i2c-bus'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/Dexter/GrovePi/Software/NodeJS/libs/grovepi.js:2:19)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/Dexter/GrovePi/Software/NodeJS/libs/sensors/base/sensor.js:3:20)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)

fivdi, 2018-07-24:
It looks like the code at https://github.com/DexterInd/GrovePi was cloned and that `npm install` was not called in the appropriate directory. I'm afraid I'm not familiar with GrovePi. Please open an issue at GrovePi as this isn't an i2c-bus issue. Perhaps someone there can help.

closed 2018-07-24 by fivdi

vkjuju, 2018-07-24:
@fivdi , thanks, I have created a new issue on : https://github.com/DexterInd/GrovePi/issues/419