fivdi/i2c-bus #56: Add promise support

fivdi, 2018-08-04:


mark-hahn, 2018-08-14:
I can't live without async/await.   Is there any kludge I can do to run i2c-bus with promises before this feature is available?

101100, 2018-08-14:
@mark-hahn Most promise libraries have methods to convert a function with a callback to a promise function.  There was some discussion of this and a fair bit of code in #29.

mark-hahn, 2018-08-14:
Thanks.  The latter code using pify looks promising.   Pify is getting 7,353,906 (!) downloads a week on npm.  That is insane.

fivdi, 2018-08-14:
Alternatively, if you're only using a few i2c-bus methods it should be possible to wrap each method with a few lines of code like [writeByte here](https://github.com/fivdi/mcp9808-temperature-sensor/blob/master/mcp9808-temperature-sensor.js#L178).

mark-hahn, 2018-08-14:
I ended up doing basically that.  Pify is so easy to use I'm just doing this for every call ...
```
  i2cWriteP = pify(bus.i2cWrite);
```
Then I can just do ...
```
 await i2cWriteP(8, 1, buf);
```

There is a catch though.  When i2cWrite is called `this` is undefined.  I'm supposed to be able to do ...
```
i2cWriteP = pify(bus.i2cWrite).bind(bus);
```
but it isn't working.  I posted an issue on the pify repo.  For now I've hacked i2cWrite to use a global `bus` instead of `this` in an i2c-bus fork.  I'm only using one bus.

You could do a release with promises by just somehow working around the `this` problem.

101100, 2018-08-14:
I think this might work:

```
i2cWriteP = pify(bus.i2cWrite.bind(bus));
```

That would bind the function before giving it to `pify` which should work unless `pify` uses some attributes that `bind` would hide.

mark-hahn, 2018-08-14:
That didn't work either.

dkebler, 2019-02-16:
Been there done that awhile ago.  There is a closed issue on this.

can use pify but you have to do it one by one.  You can't pify the whole thing.   
where is example of one method pify'd

```
read: function(args) {
    // console.log('read: address, cmd', address, cmd)
    return pify(this.i2cbus.readByte).bind(this.i2cbus)(args.address, args.cmd)
  },

I wrote an es6 class wrapper for this project.  Makes doing async/await easy.  See here.

https://git.kebler.net/UCOMmandIt/uci-i2c-bus/src/branch/master/src/bus.js

fivdi, 2019-02-16:
@dkebler Thank you for the information. I'll take a look at your implementation.

dkebler, 2019-02-16:
it's actually a class that extends my socket base class which is more than what you want but take a look at this part below in particular.  It could just extend a simple emitter class instead or just nothing at all just promise wrapper for each api call.

It has a validator because as you might remember from another issue I could not catch errors thrown by bad arguments so had so do some validation to prevent problems of uncaught errors potentially crashing my code.

```javascript

const validateArgs = function(packet) {
  let missing = []
  const ne = arg => {
    if (packet.args[arg] === undefined) missing.push(arg)
  }
  if (packet.cmd === 'scan' || packet.cmd === 'close') return {}
  ne('address')
  switch (packet.cmd) {
  case 'readRaw':
  case 'writeRaw':
    ne('length')
    ne('buffer')
    break
  case 'read':
  case 'read2':
  case 'write':
  case 'write2':
    ne('cmd')
  }
  switch (packet.cmd) {
  case 'write':
  case 'write2':
  case 'send':
    ne('byte')
  }
  if (missing.length > 0) {
    return {
      error: `following bus arguments are missing ${missing}`,
      packet: packet
    }
  }
  return {}
}

const bus_funcs = {
  scan: function() {
    return pify(this.i2cbus.scan).bind(this.i2cbus)()
  },
  close: function() {
    return pify(this.i2cbus.close).bind(this.i2cbus)()
  },
  readRaw: function(args) {
    return pify(this.i2cbus.i2cRead).bind(this.i2cbus)(
      args.address,
      args.length,
      args.buffer
    )
  },
  writeRaw: function(args) {
    return pify(this.i2cbus.i2cWrite).bind(this.i2cbus)(
      args.address,
      args.length,
      args.buffer
    )
  },
  read: function(args) {
    // console.log('read: address, cmd', address, cmd)
    return pify(this.i2cbus.readByte).bind(this.i2cbus)(args.address, args.cmd)
  },
  write: function(args) {
    // console.log('write: address, cmd, byte', args.address, args.cmd, args.byte)
    return pify(this.i2cbus.writeByte.bind(this.i2cbus))(
      args.address,
      args.cmd,
      args.byte
    )
  },
  read2: function(args) {
    return pify(this.i2cbus.readWord.bind(this.i2cbus))(args.address, args.cmd)
  },
  write2: function(args) {
    return pify(this.i2cbus.writeWord.bind(this.i2cbus))(
      args.address,
      args.cmd,
      args.byte
    )
  },
  receive: function(args) {
    // console.log('receivebyte', address)
    return pify(this.i2cbus.receiveByte.bind(this.i2cbus))(args.address)
  },
  send: function(args) {
    // console.log('sendbyte', address,byte)
    return pify(this.i2cbus.sendByte.bind(this.i2cbus))(args.address, args.byte)
  }
} //end i2c function``

dkebler, 2019-02-24:
looks like someone has already done such. https://www.npmjs.com/package/i2c-bus-promised 

Only downside might be whether they keep the i2c-bus dependency up to date.
If you maintained it that would be so.

They did the same as myself making a bus and device class
Still my version works for my use case fine and has my extra socket baggage.

I wonder if they do any validation cause as I found out passing missing or bad arguments creates issues.  Although I didn't have an issue they addressed having a queue to keep the bus from being swamped, although I thought your C code already deals with that?? so it wasn't necessary?

fivdi, 2019-02-24:
@dkebler It looks like [i2c-bus-promised](https://github.com/AlejandroHerr/i2c-bus-promised) has been superseded by [async-i2c-bus](https://github.com/AlejandroHerr/async-i2c-bus). The author of i2c-bus-promised mentions that the queue was used to avoid blocking calls. I don't know which blocking calls are being referred to. async-i2c-bus doesn't appear to have a queue. Perhaps the goal was to prevent i2c-bus-promised from depleting the Node.js / libuv thread pool when a program makes many request simultaneously without waiting for any of them to complete.

i2c-bus doesn't throttle or queue requests and it doesn't prevent a program from swamping an I2C bus. It relies on the drivers that are part of the operating system to serialize accesses to an I2C bus where necessary.

alex-page, 2019-04-06:
Do you want a PR for this @fivdi ?

fivdi, 2019-04-06:
Believe it or not I'm actually working on it but I keep getting distracted with other topics.

codebeat-nl, 2019-06-02:
I think wanting this is against the nature of the I2C bus design. You have to be careful not to screw up the I2C bus data and to confuse or malfunction the bus or other 'listeners'. The design of @dkebler is not that great, to do it right you need a queue mechanism to be sure that everything request and reply (event) is managed in the right other. 

Better is to create queue master class (server/manager) and all other 'processes' connects to this class (to receive events and updates at the same time) and to do requests. All clients needs to know what is happening to do it right. A promise is a wrong construction in this case and can introduce allot of polling overhead, it needs to be an event and can be an immediate result when doing a request when there is a 'manager'. 

Don't design it the way you want it to be, design to as close what it is possible.

fivdi, 2019-06-03:
> I think wanting this is against the nature of the I2C bus design. You have to be careful not to screw up the I2C bus data and to confuse or malfunction the bus or other 'listeners'.

@codebeat-nl I've read what you say here a few times but I'm afraid I don't understand. To help me understand it can you give an example of code that is NOT careful and screws up the I2C bus data and confuses or malfunctions the bus or other 'listeners'.

> Better is to create queue master class (server/manager) and all other 'processes' connects to this class (to receive events and updates all at the same time) and to do requests. All clients needs to know what is happening to do it right. A promise is a wrong construction in this case and can introduce allot of polling overhead, it needs to be an event and can be an immediate result when doing a request when there is a 'manager'.

I afraid I don't understand here either. I'm not sure what you mean by processes, events or updates. I'm not sure why processes should receive events and updates all at the same time. I'm not sure what a client is and why all clients need to know what is happening to do it right. I don't understand why a promise is a wrong construction in this case and can introduce allot of polling overhead. I don't understand why it needs to be an event and can be an immediate result when doing a request when there is a 'manager'. In order to understand what you mean I'd need further information. Sample code showing what the issues are would help most.

dkebler, 2019-09-12:
@fivdi I have an updated my i2c bus class.  It's not (yet) ready for public consumption (no docs/readme).  It is designed to make function calls with arguments via sockets using a json packet (depends on a currently private package) but if you/others want to look at/clone it for your purposes be my guest. https://git.kebler.net/UCOMmandIt/uci-i2c-bus

It allows one to easily use/rename any of the i2c-bus functions as promises. Further it does better validation of arguments (missing and their types/values ). It doesn't yet check proper buffer size vs length.   Of note per unvalidated arguments I have confirmed that if one promisifies the call and then captures errors one can recover from bad/missing arguments so although argument validation is preferable it's not necessary

per @codebeat-nl I added in a queue using (p-queue by Sindreshorus). If one suspects concurrency issues you can use a concurrency of 1 (default).  Otherwise some initial fiddling seems like multiple concurrency still does function calls in order without losing any "collisions on bus". In a larger application I had some issues with concurrent calls of writeByte and I hope this might solve those without having to manage the calls in my app.

All this said/provided, circa 2019, it is IMO appropriate for a maintained API to support promises.  In this way we don't have to have another package which may or may not keep up with I2c-bus releases.  Here is my suggestion to maintain backward compatibility as I have seen done in other projects.  **If the user invokes the function with a missing callback then pass them back the promise version instead**. Should only have to check to see if the last argument is a function since that is the consistent place across all the API and no other arguments are functions. As to validation of arguments let users catch errors or validate themselves similar to what I have done, no need to incorporate argument validation

fivdi, 2019-09-15:
PR #77 which has been merged (but not yet published on npm) adds support for promises to i2c-bus. I appologize for it haven taken so long for promise support to be added to i2c-bus.

fivdi, 2019-09-15:
@dkebler where do you think that argument validation is missing or could be improved in i2c-bus? To the best of my knowledge argument validation is always performed and I'm not aware of what could changed or improved.

fivdi, 2019-09-17:
Support for promises added with #77 and published on npm with i2c-bus@5.0.0

closed 2019-09-17 by fivdi