fivdi/i2c-bus #57: Odroid XU4 Support

msi-matthew, 2018-09-20:
Will this library work with the Odroid XU4?

fivdi, 2018-09-20:
I'm afraid I don't know as I've never tried it. Have you tried it?

Don't forget that GPIO pins on the XU4 operate at 1.8V but many devices support only 3.3V or 5V.

msi-matthew, 2018-09-20:
I have not. I am running into errors reading from an INA219 and am looking to see if there are alternate boards that will work with a current sensor.

`/home/pi//node_modules/i2c-bus/i2c-bus.js:355
  return i2c.readI2cBlockSync(peripheralSync(this, addr), cmd, length, buffer);
             ^

Error: , Remote I/O error
    at Bus.readI2cBlockSync (/home/pi/node_modules/i2c-bus/i2c-bus.js:355:14)
    at Ina219.readRegister (/home/pi/node_modules/ina219/ina219.js:184:12)
    at Ina219.getShuntVoltage_raw (/home/pi/node_modules/ina219/ina219.js:448:7)
    at Ina219.getShuntVoltage_mV (/home/pi/node_modules/ina219/ina219.js:500:7)
    at Timeout.setInterval [as _onTimeout] (/home/pi/test.js:44:12)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)`

msi-matthew, 2018-09-20:
I have tried editing `/boot/config.txt` with `dtparam=i2c_arm_baudrate=50000` and get the same error.

Setting `dtoverlay=i2c-gpio,bus=3` gave a different error...

fivdi, 2018-09-20:
As you have already noticed the issue is likely to be related to [this one](https://github.com/fivdi/i2c-bus/issues/36).

While trying with `dtoverlay=i2c-gpio,bus=3` was the call to [ina219.init(address, device)](https://github.com/brettmarl/node-ina219#ina219initaddress-device) modified to pass `3` as the device?
While trying with `dtoverlay=i2c-gpio,bus=3` was the ina219 connected to GPIO23 (SDA) and GPIO24 (SCL)?
While trying with `dtoverlay=i2c-gpio,bus=3` what error was shown?

msi-matthew, 2018-09-20:
While trying with dtoverlay=i2c-gpio,bus=3 was the call to ina219.init(address, device) modified to pass 3 as the device?

No, it was not, should I also pass `0x40` for the address or would null suffice?

While trying with dtoverlay=i2c-gpio,bus=3 was the ina219 connected to GPIO23 (SDA) and GPIO24 (SCL)?

Yes.

While trying with dtoverlay=i2c-gpio,bus=3 what error was shown?

Let me re-run the test. I'll post an update with the log.

msi-matthew, 2018-09-20:
With `dtoverlay=i2c-gpio,bus=3` set in `boot/config.txt` 
and `ina219.init()` I get the following error:

```
/home/pi/node_modules/i2c-bus/i2c-bus.js:520
  i2c.writeI2cBlockSync(peripheralSync(this, addr), cmd, length, buffer);
      ^

Error: , Remote I/O error
    at Bus.writeI2cBlockSync (/home/pi/node_modules/i2c-bus/i2c-bus.js:520:7)
    at Ina219.writeRegister (/home/pi/node_modules/ina219/ina219.js:162:12)
    at Object.<anonymous> (/home/pi/test.js:22:8)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:191:16)

msi-matthew, 2018-09-20:
With dtoverlay=i2c-gpio,bus=3 set in boot/config.txt
and ina219.init('INA219_ADDRESS', 3) I get the following error:

```
/home/pi//node_modules/i2c-bus/i2c-bus.js:53
    throw new Error('Invalid I2C address ' + addr);
    ^

Error: Invalid I2C address INA219_ADDRESS
    at checkAddress (/home/pi/node_modules/i2c-bus/i2c-bus.js:53:11)
    at Bus.writeI2cBlockSync (/home/pi/node_modules/i2c-bus/i2c-bus.js:516:3)
    at Ina219.writeRegister (/home/pi/node_modules/ina219/ina219.js:162:12)
    at Object.<anonymous> (/home/test.js:22:8)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)

msi-matthew, 2018-09-20:
With `dtoverlay=i2c-gpio,bus=3` set in `boot/config.txt`
and `ina219.init('0x40', 3)` I get the following error::

```
/home/pi//node_modules/i2c-bus/i2c-bus.js:53
    throw new Error('Invalid I2C address ' + addr);
    ^

Error: Invalid I2C address 0x40
    at checkAddress (/home/pi/node_modules/i2c-bus/i2c-bus.js:53:11)
    at Bus.writeI2cBlockSync (/home/pi/node_modules/i2c-bus/i2c-bus.js:516:3)
    at Ina219.writeRegister (/home/pi/node_modules/ina219/ina219.js:162:12)
    at Object.<anonymous> (/home/test.js:22:8)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)

fivdi, 2018-09-20:
Try with:
```js
ina219.init(0x40, 3);
```

msi-matthew, 2018-09-20:
With dtoverlay=i2c-gpio,bus=3 set in boot/config.txt
and ina219.init(0x40, 3) I get the following error:

/home/pi/node_modules/i2c-bus/i2c-bus.js:520
  i2c.writeI2cBlockSync(peripheralSync(this, addr), cmd, length, buffer);
      ^

Error: , Remote I/O error
    at Bus.writeI2cBlockSync (/home/pi/node_modules/i2c-bus/i2c-bus.js:520:7)
    at Ina219.writeRegister (/home/pi/node_modules/ina219/ina219.js:162:12)
    at Object.<anonymous> (/home/pi/test.js:22:8)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:191:16)

msi-matthew, 2018-09-20:
A list of addresses from `ina219.js`

```
// ===========================================================================
//   I2C ADDRESS/BITS
// ==========================================================================
var INA219_ADDRESS                         	= 0x40    ; // 1000000 (A0+A1=GND)
var INA219_ADDRESS_A0                      	= 0x41    ; 
var INA219_ADDRESS_A1                      	= 0x44    ; 
var INA219_ADDRESS_A0_A1                   	= 0x45    ;

var INA219_READ 							= 0x01; 
// ===========================================================================
//    CONFIG REGISTER (R/W)
// ===========================================================================
var INA219_REG_CONFIG                      	= 0x00;

// ===========================================================================
var INA219_CONFIG_RESET                    	= 0x8000  ; // Reset Bit
var INA219_CONFIG_BVOLTAGERANGE_MASK       	= 0x2000  ; // Bus Voltage Range Mask
var INA219_CONFIG_BVOLTAGERANGE_16V        	= 0x0000  ; // 0-16V Range
var INA219_CONFIG_BVOLTAGERANGE_32V        	= 0x2000  ; // 0-32V Range

var INA219_CONFIG_GAIN_MASK                	= 0x1800  ; // Gain Mask
var INA219_CONFIG_GAIN_1_40MV              	= 0x0000  ; // Gain 1, 40mV Range
var INA219_CONFIG_GAIN_2_80MV              	= 0x0800  ; // Gain 2, 80mV Range
var INA219_CONFIG_GAIN_4_160MV             	= 0x1000  ; // Gain 4, 160mV Range
var INA219_CONFIG_GAIN_8_320MV             	= 0x1800  ; // Gain 8, 320mV Range

var INA219_CONFIG_BADCRES_MASK             	= 0x0780  ; // Bus ADC Resolution Mask
var INA219_CONFIG_BADCRES_9BIT             	= 0x0080  ; // 9-bit bus res = 0..511
var INA219_CONFIG_BADCRES_10BIT            	= 0x0100  ; // 10-bit bus res = 0..1023
var INA219_CONFIG_BADCRES_11BIT            	= 0x0200  ; // 11-bit bus res = 0..2047
var INA219_CONFIG_BADCRES_12BIT            	= 0x0400  ; // 12-bit bus res = 0..4097

var INA219_CONFIG_SADCRES_MASK             	= 0x0078  ; // Shunt ADC Resolution and Averaging Mask
var INA219_CONFIG_SADCRES_9BIT_1S_84US     	= 0x0000  ; // 1 x 9-bit shunt sample
var INA219_CONFIG_SADCRES_10BIT_1S_148US   	= 0x0008  ; // 1 x 10-bit shunt sample
var INA219_CONFIG_SADCRES_11BIT_1S_276US   	= 0x0010  ; // 1 x 11-bit shunt sample
var INA219_CONFIG_SADCRES_12BIT_1S_532US   	= 0x0018  ; // 1 x 12-bit shunt sample
var INA219_CONFIG_SADCRES_12BIT_2S_1060US  	= 0x0048  ; // 2 x 12-bit shunt samples averaged together
var INA219_CONFIG_SADCRES_12BIT_4S_2130US  	= 0x0050  ; // 4 x 12-bit shunt samples averaged together
var INA219_CONFIG_SADCRES_12BIT_8S_4260US  	= 0x0058  ; // 8 x 12-bit shunt samples averaged together
var INA219_CONFIG_SADCRES_12BIT_16S_8510US 	= 0x0060  ; // 16 x 12-bit shunt samples averaged together
var INA219_CONFIG_SADCRES_12BIT_32S_17MS   	= 0x0068  ; // 32 x 12-bit shunt samples averaged together
var INA219_CONFIG_SADCRES_12BIT_64S_34MS   	= 0x0070  ; // 64 x 12-bit shunt samples averaged together
var INA219_CONFIG_SADCRES_12BIT_128S_69MS  	= 0x0078  ; // 128 x 12-bit shunt samples averaged together

var INA219_CONFIG_MODE_MASK                	= 0x0007  ; // Operating Mode Mask
var INA219_CONFIG_MODE_POWERDOWN 			= 0x0000;
var INA219_CONFIG_MODE_SVOLT_TRIGGERED 	 	= 0x0001;
var INA219_CONFIG_MODE_BVOLT_TRIGGERED 	 	= 0x0002;
var INA219_CONFIG_MODE_SANDBVOLT_TRIGGERED 	= 0x0003;
var INA219_CONFIG_MODE_ADCOFF 			 	= 0x0004;
var INA219_CONFIG_MODE_SVOLT_CONTINUOUS 	= 0x0005;
var INA219_CONFIG_MODE_BVOLT_CONTINUOUS 	= 0x0006;
var INA219_CONFIG_MODE_SANDBVOLT_CONTINUOUS	= 0x0007;

// ===========================================================================
//   SHUNT VOLTAGE REGISTER (R)
// ===========================================================================
var INA219_REG_SHUNTVOLTAGE                	= 0x01
// ===========================================================================

// ===========================================================================
//   BUS VOLTAGE REGISTER (R)
// ===========================================================================
var INA219_REG_BUSVOLTAGE                  	= 0x02
// ===========================================================================

// ===========================================================================
//   POWER REGISTER (R)
// ===========================================================================
var INA219_REG_POWER                       	= 0x03
// ===========================================================================

// ==========================================================================
//    CURRENT REGISTER (R)
// ===========================================================================
var INA219_REG_CURRENT                     	= 0x04
// ===========================================================================

// ===========================================================================
//    CALIBRATION REGISTER (R/W)
// ===========================================================================
var INA219_REG_CALIBRATION                 	= 0x05
// ===========================================================================

fivdi, 2018-09-20:
What's the output of the command
```
i2cdetect -y -r 3
```

msi-matthew, 2018-09-20:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

fivdi, 2018-09-20:
The INA219 is not correctly connected to the Raspberry Pi.

msi-matthew, 2018-09-20:
I have SDA connected to pin 23 and SCL connected to pin 24.

Should I be using the map from this diagram? (https://www.raspberrypi-spy.co.uk/2014/07/raspberry-pi-b-gpio-header-details-and-pinout/)

msi-matthew, 2018-09-20:
I should note that I'm currently using a Raspberry Pi, the original title may be a bit misleading at this point.

fivdi, 2018-09-20:
SDA should be on GPIO23 and SCL should be on GPIO24 which are pins 16 and 18 on the GPIO header respectively.

msi-matthew, 2018-09-20:
Pins 16 and 18 are flashing on a breakout relay when I run `i2cdetect -y -r 3`

msi-matthew, 2018-09-20:
Plugged into 16 and 18 I get 

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   ```

Which produces the same error with `ina219.init(0x40, 3);`

 get i2c.writeI2cBlockSync(peripheralSync(this, addr), cmd, length, buffer);

fivdi, 2018-09-22:
As can be seen in the output of `i2cdetect -y -r 3`, the ina219 at address 0x40 is now correctly connected to I2C bus 3 (/dev/i2c-3).

Unfortunately there's a problem with [node-ina219](https://github.com/brettmarl/node-ina219), the Node.js module being used to access the ina219.

The version of node-ina219 published on npm isn't the same as the version published on github. The init method published on npm looks like this:

```js
Ina219.prototype.init = function (address, device) {

	// defaults
	address = typeof address !== 'undefined' ? address : INA219_ADDRESS;
	device = typeof device !== 'undefined' ? device : '/dev/i2c-1';
	
	this.log("init:: " + address + " | " + device)
	this.currentDivider_mA = 0;
	this.powerDivider_mW = 0;
	this.calValue = 0;
	this.address = address;
	
	this.wire = i2c.openSync(1);	// 1 == /dev/ic2-1
	
	//this.wire = new i2c(address, { device: device }); // point to your i2c address, debug provides REPL interface
}
```

and the init method published on github looks like [this](https://github.com/brettmarl/node-ina219/blob/master/ina219.js#L120-L133):
```js
Ina219.prototype.init = function (address, busNumber) {

	// defaults
	address = typeof address !== 'undefined' ? address : INA219_ADDRESS;
	busNumber = typeof busNumber !== 'undefined' ? busNumber : 1;
	
	this.log("init:: " + address + " | " + busNumber)
	this.currentDivider_mA = 0;
	this.powerDivider_mW = 0;
	this.calValue = 0;
	this.address = address;
	
	this.wire = i2c.openSync(busNumber);
}
```
The init method published on npm only supports I2C bus 1. The init method published on github supports any I2C bus number. I didn't spot this earlier on.

To proceed forward there are a number of options.

1. Find the file `ina219.js` on your Raspberry Pi and replace its content with the content of [ina219.js](https://github.com/brettmarl/node-ina219/blob/master/ina219.js).
2. Create an issue at [node-ina219](https://github.com/brettmarl/node-ina219) asking the author to publish the latest version of node-ina219 in order to make it possible to use I2C bus 3.
3. Try using I2C bus 1 again and ensure that the ina219 is correctly connected. Perhaps the ina219 was not correctly connected to I2C bus 1. Use `i2cdetect -y -r 1` to verify that the ina219 is connected to I2C bus 1.

msi-matthew, 2018-09-22:
Awesome lead! 

My test hardware is off premises at the moment. I will continue this path on Monday...

Thank you!!

msi-matthew, 2018-09-27:
I was able to test the Github module this week, and even with bus 3 configured it throws the following error occasionally:

```
/home/pi/Documents/backend/node_modules/i2c-bus/i2c-bus.js:355
  return i2c.readI2cBlockSync(peripheralSync(this, addr), cmd, length, buffer);
             ^

Error: EIO, Input/output error
    at Bus.readI2cBlockSync (/home/pi/Documents/backend/node_modules/i2c-bus/i2c-bus.js:355:14)
    at Ina219.readRegister (/home/pi/Documents/backend/node_modules/ina219/ina219.js:181:12)
    at Ina219.getShuntVoltage_raw (/home/pi/Documents/backend/node_modules/ina219/ina219.js:445:7)
    at Ina219.getShuntVoltage_mV (/home/pi/Documents/backend/node_modules/ina219/ina219.js:497:7)
    at Timeout.setInterval [as _onTimeout] (/home/pi/Documents/backend/k.js:44:12)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)

fivdi, 2018-09-27:
"EIO, Input/output error" and "Remote I/O error" mean more or less the same thing.

If the problem is still occurring on bus 3 and the ina219 is connected up correctly then I think the only thing that can be done is to catch errors and if they are "EIO, Input/output errors" or "Remote I/O errors" retry the operation. This is assuming that the "EIO, Input/output error" doesn't always occur.

It's not going to be possible to modify i2c-bus to resolve this issue.

msi-matthew, 2018-09-27:
Or this one?

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 31 32 33 34 35 36 37 -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 51 52 53 54 55 56 57 58 -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

msi-matthew, 2018-09-27:
> If the problem is still occurring on bus 3 and the ina219 is connected up correctly then I think the only thing that can be done is to catch errors and if they are "EIO, Input/output errors" or [...]

This is something I'd like to pursue. You sent a link earlier.  

Is the idea to edit the read function in the `node-ina219` module?

msi-matthew, 2018-09-27:
Or this one?

``` 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

msi-matthew, 2018-09-27:
I'm doing some more tests on an Orange Pi and am getting the following error: 

```
/home/msi/Documents/backend/node_modules/i2c-bus/i2c-bus.js:520
  i2c.writeI2cBlockSync(peripheralSync(this, addr), cmd, length, buffer);
      ^

Error: EAGAIN, Resource temporarily unavailable
    at Bus.writeI2cBlockSync (/home/msi/Documents/backend/node_modules/i2c-bus/i2c-bus.js:520:7)
    at Ina219.writeRegister (/home/msi/Documents/backend/node_modules/ina219/ina219.js:159:12)
    at Object.<anonymous> (/home/msi/Documents/backend/orange.js:24:8)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:188:16)

fivdi, 2018-09-27:
If the output of running `i2cdetect -y -r 3` varies from run to run like this ...
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --        
```
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 31 32 33 34 35 36 37 -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 51 52 53 54 55 56 57 58 -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```
... then there's something wrong with the hardware setup. You'll need to figure out what's going wrong with the hardware. The output of `i2cdetect -y -r 3` should always be the same.

If I had to guess I'd say EAGAIN is a result of a faulty hardware setup.

msi-matthew, 2018-09-27:
OK. 

Those were results booting into various I2C configurations (enable/disable options - IC20, IC21, IC23, etc.) from armbian-config on an Orange Pi.

fivdi, 2018-09-27:
This issue started off talking about the Odroid XU4, moved over to the Raspberry Pi and now has moved to a Orange Pi. The move to the Orange Pi was mentioned after the move. Things are getting too difficult to follow.

fivdi, 2018-09-28:
> This is something I'd like to pursue. You sent a link earlier.
>
> Is the idea to edit the read function in the node-ina219 module?

The idea would be to add proper error handling to either node-ina219 or your application. Perhaps it's better to add it to your application. Any call to an i2c-bus function can fail. Currently your application isn't handling these errors so the application crashes.

I'm going to go ahead and close this issue as there is TODO for i2c-bus.

closed 2018-09-28 by fivdi