fivdi/i2c-bus #58: How to Handle Errors

msi-matthew, 2018-10-03:
When running this code I get the following error:

> /home/pi/Documents/node_modules/i2c-bus/i2c-bus.js:355 return
> i2c.readI2cBlockSync(peripheralSync(this, addr), cmd, length, buffer);
> Error: , Communication error on send

How can I catch it and continue to run code?

    const ina219 = require('ina219')
    const DELAY_MS = 15

    try {
      let getVoltage = ina219.calibrate32V1A(() => {
        let int = setInterval(() => {
          ina219.getShuntVoltage_mV((volts) => {
            console.log(volts)
          })
        }, DELAY_MS)
      })
    }
    catch (error) {
      console.log(error)
    }

fivdi, 2018-10-03:
Please post the complete stack trace of the error.

msi-matthew, 2018-10-03:
/home/orangepi/Documents/backend/node_modules/i2c-bus/i2c-bus.js:355
      return i2c.readI2cBlockSync(peripheralSync(this, addr), cmd, length, buffer);
                 ^

    Error: , Communication error on send
        at Bus.readI2cBlockSync (/home/orangepi/Documents/backend/node_modules/i2c-bus/i2c-bus.js:355:14)
        at Ina219.readRegister 
    (/home/orangepi/Documents/backend/node_modules/ina219/ina219.js:181:12)
        at Ina219.getShuntVoltage_raw 
    (/home/orangepi/Documents/backend/node_modules/ina219/ina219.js:445:7)
        at Ina219.getShuntVoltage_mV 
    (/home/orangepi/Documents/backend/node_modules/ina219/ina219.js:497:7)
        at Timeout.setInterval [as _onTimeout] (/home/orangepi/Documents/backend/switch-test.js:47:14)
        at ontimeout (timers.js:498:11)
        at tryOnTimeout (timers.js:323:5)
        at Timer.listOnTimeout (timers.js:290:5)

fivdi, 2018-10-03:
Is it the case that the ina219 never works or is the the case that it occasionally doesn't work?

msi-matthew, 2018-10-03:
It occasionally doesn't work.

It will throw an error every few hundred or so reads.

fivdi, 2018-10-03:
What is on line 47 of switch-test.js?

msi-matthew, 2018-10-03:
ina219.getShuntVoltage_mV((volts) => {

fivdi, 2018-10-03:
The code in the ina219 module invokes callbacks synchronously which is a little strange. `getShuntVoltage_mV` looks like an asynchronous method as it has a callback but it's actually a synchronous method.

Try the following, it should ignore errors thrown by `getShuntVoltage_mV`. It doesn't handle errors thrown by `calibrate32V1A` and would need to be extended to handle  them. I'll leave this as an exercise.
```js
const ina219 = require('ina219')
const DELAY_MS = 15

let getVoltage = ina219.calibrate32V1A(() => {
  let int = setInterval(() => {
    try {
      ina219.getShuntVoltage_mV((volts) => {
        console.log(volts)
      })
    } catch (ignore) {
      console.log('ignoring error thrown by getShuntVoltage_mV')
    }
  }, DELAY_MS)
})
```

msi-matthew, 2018-10-03:
Very charming.

Thank you!!

fivdi, 2018-10-03:
You're welcome :)

closed 2018-10-03 by fivdi