fivdi/i2c-bus #61: Cannot install i2c-bus >= 4.0.3 on macOS

Ks89, 2018-11-13:
macOS Mojave with all updates installed
nodeJS: 8.11.1
npm: 6.4.1

Version 4.0.2 is working, but when I switch to 4.0.3/4.0.5 it appears this error:

Command failed: npm install --save i2c-bus@4.0.5 --color=always
In file included from ../src/i2c.cc:17:
../src/./i2c-dev.h:26:10: fatal error: 'linux/types.h' file not found
#include <linux/types.h>
         ^~~~~~~~~~~~~~~
1 error generated.
make: *** [Release/obj.target/i2c/src/i2c.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Darwin 18.2.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/stefanocappa/git/easykon/server/node_modules/i2c-bus
gyp ERR! node -v v8.11.1
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN ajv-keywords@2.1.1 requires a peer of ajv@^5.0.0 but none is installed. You must install peer dependencies yourself.

fivdi, 2018-11-18:
Sorry for any inconveniences caused here. Modifications to `bindings.gyp` in i2c-bus@4.0.3 broke the install on macOS. Everything should work again with i2c-bus@4.0.6 which has been published on npm. Please give i2c-bus@4.0.6 a try to see if it works. I would be very grateful if you could let me know if it works as I don't have a Mac to test with.

Ks89, 2018-11-18:
Ok, tomorrow I'll try.
thank u.

Ks89, 2018-11-19:
It's working. thank u for the support

closed 2018-11-19 by Ks89

fivdi, 2018-11-23:
@Ks89 thanks for the feedback, it's greatly appreciated.