fivdi/i2c-bus #66: Write, and then read, multiple number of Bytes

captain1991cv, 2019-03-09:
Dear @fivdi 
In your documentation, you have mentioned that block functions (e.g: ```bus.readI2cBlock(addr, cmd, length, buffer, cb)```) are ```not defined by the SMBus specification```. 
According to SMbus specification [(here)](http://ww1.microchip.com/downloads/en/AppNotes/doc2583.pdf) The “Block write/read” protocol can be used to send/receive up to 32 bytes of data to/from a slave device, something like this:

[image]

what is the difference between SMbus Block “write/read” and this library's functions? do you send ```Byte Count``` in above picture at the beginning?
I need to read multiple Bytes from my slave (e.g 3 Bytes or 9 Bytes), in other words, i need my slave to send me its data until i send the NACK. something like this:
[image]

how can i do that? is ```bus.readI2cBlock(addr, cmd, length, buffer, cb)``` what I'm searching for?

fivdi, 2019-03-09:
Hi @captain1991cv,

> In your documentation, you have mentioned that block functions (e.g: `bus.readI2cBlock(addr, cmd, length, buffer, cb)`) are `not defined by the SMBus specification`. 

Yes, that is correct, `readI2cBlock` and `writeI2cBlock` are not defined by the SMBus specification. They are however supported by the SMBus layer on Linux. Here is what the [Linux kernel SMBus Protocol Documentation](https://www.kernel.org/doc/Documentation/i2c/smbus-protocol) has to say about these I2C block transactions:
```
I2C Block Transactions
======================

The following I2C block transactions are supported by the
SMBus layer and are described here for completeness.
They are *NOT* defined by the SMBus specification.

I2C block transactions do not limit the number of bytes transferred
but the SMBus layer places a limit of 32 bytes.

I2C Block Read:  i2c_smbus_read_i2c_block_data()
================================================

This command reads a block of bytes from a device, from a 
designated register that is specified through the Comm byte.

S Addr Wr [A] Comm [A] 
           S Addr Rd [A] [Data] A [Data] A ... A [Data] NA P

Functionality flag: I2C_FUNC_SMBUS_READ_I2C_BLOCK

I2C Block Write:  i2c_smbus_write_i2c_block_data()
==================================================

The opposite of the Block Read command, this writes bytes to 
a device, to a designated register that is specified through the
Comm byte. Note that command lengths of 0, 2, or more bytes are
supported as they are indistinguishable from data.

S Addr Wr [A] Comm [A] Data [A] Data [A] ... [A] Data [A] P

Functionality flag: I2C_FUNC_SMBUS_WRITE_I2C_BLOCK
```
`i2c_smbus_read_i2c_block_data` corresponds to `readI2cBlock` and `i2c_smbus_write_i2c_block_data` corresponds to `writeI2cBlock`.

Looking at the second image you posted above, `readI2cBlock` looks like it will do what's required. Note that `readI2cBlock` needs to be told how many bytes to read with the `length` parameter.

> According to SMbus specification [(here)](http://ww1.microchip.com/downloads/en/AppNotes/doc2583.pdf) The “Block write/read” protocol can be used to send/receive up to 32 bytes of data to/from a slave device, ...

i2c-bus does actually have untested and undocumented support for the SMBus block read and block write protocols. The corresponding methods are called `readBlock` and `writeBlock`. I have never tested these methods as I haven't got any hardware that (fully) supports them. Running the command `i2cdetect -F 1` on a Raspberry Pi displays the following:
```
pi@raspberrypi:~ $ i2cdetect -F 1
Functionalities implemented by /dev/i2c-1:
I2C                              yes
SMBus Quick Command              yes
SMBus Send Byte                  yes
SMBus Receive Byte               yes
SMBus Write Byte                 yes
SMBus Read Byte                  yes
SMBus Write Word                 yes
SMBus Read Word                  yes
SMBus Process Call               yes
SMBus Block Write                yes
SMBus Block Read                 no
SMBus Block Process Call         no
SMBus PEC                        yes
I2C Block Write                  yes
I2C Block Read                   yes
```
As can be seen, `SMBus block write` is supported on a raspberry Pi but `SMBus block read` isn't.

> what is the difference between SMbus Block “write/read” and this library's functions? do you send Byte Count in above picture at the beginning?

The main difference, as you have noticed, is that the SMBus block read/write protocols include a byte count in the data transferred. `readI2cBlock` and `writeI2cBlock` don't include a byte count in the data transferred. Another difference is that `i2c_smbus_read_block_data` (readBlock) doesn't need to be told how many bytes it should read, but `i2c_smbus_read_i2c_block_data` (readI2cBlock) does need t be told.

>is ```bus.readI2cBlock(addr, cmd, length, buffer, cb)``` what I'm searching for?

I'd say it is.

captain1991cv, 2019-03-09:
Thank you for your respond and your support, The answer was so clear, so i close this issue.

closed 2019-03-09 by captain1991cv