fivdi/i2c-bus #70: 16 bit register address

evdokimosk, 2019-04-16:
Hello fivdi and contributors,

thank you for the i2c code. I am working on reading a VL53L1X from Raspberry by using node.
The issue I came across is that the VL53L1X uses 16bit to address register (the cmd that appears in you functions). Your code is more than helpful when I try to write on register as I send the MSB qnd the LSB of the address as the first two elements of the buffer, like the following example:

var buffer = new Buffer ([0x00, 0x01, newAddress]);
i2c1.i2cWriteSync(this.address, 3, buffer);
where I write the newAddress to the register at 0x0001

the problem is when I try to read the address 0x0001 using the i2c1.readByteSync

Any help would be really appreciated,

closed 2019-04-17 by evdokimosk

williamkapke, 2019-10-02:
Interestingly, I hit this same problem with my vl53l1x development.

I managed to hack it like this...
```js
const i2c = require('i2c-bus')
i2c.openPromisified(1).then(async (bus1) => {
  const register = 0x010f
  console.log(await bus1.writeByte(0x29, (register >> 8) & 0xFF, register & 0xFF))
  console.log((await bus1.receiveByte(0x29)).toString(16))
})
.catch(console.error)
```

This checks the model id and returns `ea` as expected. It works because that function sends 2 bytes total- which is what we need to set the read index on the device. So, I used the `cmd` param for the MSBs and then the `byte` param for the LSBs and it works.

From there, we just need to `receiveByte` at the read index & violà!

[image]

fivdi, 2019-10-02:
This might do it too:
```js
const i2c = require('i2c-bus')
i2c.openPromisified(1).then(async (bus1) => {
  const register = 0x010f
  const wbuf = Buffer.from([(register >> 8) & 0xFF, register & 0xFF]);
  console.log(await bus1.i2cWrite(0x29, wbuf.length, wbuf))
  console.log((await bus1.receiveByte(0x29)).toString(16))
})
.catch(console.error)
```
Or this:
```js
const i2c = require('i2c-bus')
i2c.openPromisified(1).then(async (bus1) => {
  const register = 0x010f
  const wbuf = Buffer.from([(register >> 8) & 0xFF, register & 0xFF])
  console.log(await bus1.i2cWrite(0x29, wbuf.length, wbuf))
  const rbuf = Buffer.alloc(1)
  console.log((await bus1.i2cRead(0x29, rbuf.length, rbuf)).buffer[0].toString(16))
})
.catch(console.error)
```