fivdi/i2c-bus #71: Node 12.0.0 failure

jgsrvey955, 2019-04-24:
After installing Node v12.0.0 there is an error for Int32Val that i could catch during rebuild. After reverting back to 11.14.0 i noticed that during rebuild the setup states that it is V8_Depreciated

fivdi, 2019-04-24:
i2c-bus installs and functions correctly for me with both Node 11.14.0 and Node 12.0.0.

Here's what I see with Node 12.0.0:
```
pi@raspberrypi:~/i2c-bus $ node -v
v12.0.0
pi@raspberrypi:~/i2c-bus $ rm -rf *
pi@raspberrypi:~/i2c-bus $ npm i i2c-bus

> i2c-bus@4.0.9 install /home/pi/i2c-bus/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory '/home/pi/i2c-bus/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/home/pi/i2c-bus/node_modules/i2c-bus/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/i2c-bus/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/i2c-bus/package.json'
npm WARN i2c-bus No description
npm WARN i2c-bus No repository field.
npm WARN i2c-bus No README data
npm WARN i2c-bus No license field.

+ i2c-bus@4.0.9
added 4 packages from 9 contributors and audited 4 packages in 17.902s
found 0 vulnerabilities

pi@raspberrypi:~/i2c-bus $ cd node_modules/i2c-bus/example/
pi@raspberrypi:~/i2c-bus/node_modules/i2c-bus/example $ node i2cscan.js 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- 1e --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- 39 -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --                        
pi@raspberrypi:~/i2c-bus/node_modules/i2c-bus/example $ 
```

Here's what I see with Node 11.14.0:
```
pi@raspberrypi:~/i2c-bus $ node -v
v11.14.0
pi@raspberrypi:~/i2c-bus $ rm -rf *
pi@raspberrypi:~/i2c-bus $ npm i i2c-bus

> i2c-bus@4.0.9 install /home/pi/i2c-bus/node_modules/i2c-bus
> node-gyp rebuild

make: Entering directory '/home/pi/i2c-bus/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/home/pi/i2c-bus/node_modules/i2c-bus/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/i2c-bus/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/i2c-bus/package.json'
npm WARN i2c-bus No description
npm WARN i2c-bus No repository field.
npm WARN i2c-bus No README data
npm WARN i2c-bus No license field.

+ i2c-bus@4.0.9
added 4 packages from 9 contributors and audited 4 packages in 16.092s
found 0 vulnerabilities

pi@raspberrypi:~/i2c-bus $ cd node_modules/i2c-bus/example/
pi@raspberrypi:~/i2c-bus/node_modules/i2c-bus/example $ node i2cscan.js 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- 1e --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- 39 -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --                        
pi@raspberrypi:~/i2c-bus/node_modules/i2c-bus/example $ 
```
Please describe the exact steps needed to reproduce the error.

fivdi, 2019-04-27:
The [Travis build](https://travis-ci.org/fivdi/i2c-bus/builds/525170385) succeeds with Node 4, 6, 8, 10, 11 and 12.

jgarvey955, 2019-04-27:
I found what's going on. Two of my other modules have 1.2.5 as dependency. Even if I install 4.0.9 first they try to install 1.2.5 which fails in node 12. How can I change this?

fivdi, 2019-04-27:
In order to use Node 12 the other two modules will need to upgrade their i2c-bus dependency to 4.0.9.

closed 2019-04-27 by fivdi