fivdi/i2c-bus #73: Incorrect read values

aidv, 2019-04-30:
I'm reading three bytes and my NodeJS code reads `3, 192, 144` in decimal while my logic analyzer reads `0x03, 0xC0, 0x10` which translates to `3, 192, 16`.

Why is the last byte misread as `144` when it should in fact be read as `16`?

Additional info:
This is on a RPi2B on device i2c@10 software i2c device, no pullups, straight from RPi to i2c device.

fivdi, 2019-05-01:
> Why is the last byte misread as 144 when it should in fact be read as 16?

Perhaps it's because there are no pullup resistors. Please add pullup resistors to see if it resolves the issue.

> As you can see at some points the reads are REALLY OFF especially at address 69... What?!?! 🤔

They may look very different in decimal but in binary they look quite similar. Only 2 of the 24 bits are incorrect.
```
69 (Logic analyser decimal):   3 64 0
69 (Node readout decimal):   131 80 0

69 (Logic analyser hex): 0x03 0x40 0x00
69 (Node readout hex):   0x83 0x50 0x00

69 (Logic analyser binary): 00000011 01000000 00000000
69 (Node readout binary):   10000011 01010000 00000000
```
If all the numbers are converted to binary the same pattern can be seen. Node.js is reading some bits as 0 while the logic analyser reads them as 1. It's never the other way around in the data shown.

Does it work correctly if the hardware I2C bus is used? The Raspberry Pi has built in pullup resistors for the hardware I2C bus.

What type of I2C device is being used?

aidv, 2019-05-01:
> Please add pullup resistors
Any suggestion on what values to use?
I later tried pullups too, 4.4k. Same issue.

I also tried using a level shifter and the slave device ground and 5v as reference.

I will test the hardware i2c later today.

I have also tried using an Arduino Mega 2560 iwth internal pullups ON connected directly to the 5v i2c of the slave.

> What type of I2C device is being used?
It's a 5v i2c device from the 90's. More specifically; a fader automation system for music studios.

Any other suggestions that I could try once I get back to the lab?

fivdi, 2019-05-01:
> Any suggestion on what values to use?

The built in pullup resistors for the hardware I2C bus on the Raspberry Pi are 1.8k resistors.

> I later tried pullups too, 4.4k. Same issue.

4.4k should be ok.

> Also, should the pullup connect to 3.3v or 5v?

The pullups should only be connected to 3.3V. They should NOT be connected to 5V. The pins on the Raspberry Pi are NOT 5V tolerant,

> Any other suggestions that I could try once I get back to the lab?

 - Is the I2C device a 5V device or a 3.3V device? Or does it support both voltages? What voltage is it being powered with?
 - Do the Raspberry Pi and I2C device have a common ground?  
 - When the I2C device is powered and without pullup resistors connected, measure the voltage between GND on the Raspberry Pi and SDA/SCL on the I2C device. If the measured voltage is not 0V, i.e., if it's approximately 3.3V or approximately 5V (depending on how the device is being powered) there are most likely pullup resistors on the device. If the measured voltage is 5V then it's not going to work correctly without logic level shifters as the Raspberry Pi is not 5V tolerant. Note that bi-directional rather than uni-directional logic level shifters will be needed
 - Were bi-directional logic level shifters used in the test mentioned in the above post?
 - How long are the wires connecting the Raspberry Pi to the I2C device? Keep them as short as possible.
 - The [i2c-gpio overlay](https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L937-L952) in `/boot/config.txt` has a parameter called `i2c_gpio_delay_us` which can be used to configure the software I2C bus speed. The default value for `i2c_gpio_delay_us` is 2 which results in a bus speed of ~100kHZ. Does increasing the value of `i2c_gpio_delay_us` in order to decrease the bus speed resolve the issue? Note that if `i2c_gpio_delay_us` is modified it may be necessary to reboot the Pi for the change to take effect.

aidv, 2019-05-01:
> Is the I2C device a 5V device or a 3.3V device? Or does it support both voltages? What voltage is it being powered with?

It’s a 5v device with 5v i2c bus. It’s powered with 5v.

> Do the Raspberry Pi and I2C device have a common ground?

No, they are powered by two different power supplies

> ...bi-directional rather than uni-directional logic level shifters will be needed

I’m not sure if the logic level shifter that I’m using currently is bi-directional, however the manufacturer describes it as being i2c compatible. It’s described as 3v to 5v and not the other way around. One big note: I had to reverse the level shifter or else it wouldn’t work. Now it kind of works, just that the readouts are incorrect.

> How long are the wires connecting the Raspberry Pi to the I2C device?

As long as the original system’s setup, which will guarantee to be a minimum of 1 meter but not longer than 2 meters.

> ...parameter called i2c_gpio_delay_us...

I know for a fact based on the slave device MCU datasheet that the i2c is expected to run at 100kHz. Is there a range of delays I should try?

Additional information: Also, I have noticed that sometimes tiny spikes on the SDA signal occurs when i analyze the signal at higher speeds. What could cause this issue?

fivdi, 2019-05-01:
>>Do the Raspberry Pi and I2C device have a common ground?

>No, they are powered by two different power supplies

GND/0V on the Raspberry Pi and GND/0V on the other system need to be connected to each other.

>>How long are the wires connecting the Raspberry Pi to the I2C device?

>As long as the original system’s setup, which will guarantee to be a minimum of 1 meter but not longer than 2 meters.

That's not extremely long but it's fairly long. The following comment was taked from [here](http://www.mosaic-industries.com/embedded-systems/sbc-single-board-computers/freescale-hcs12-9s12-c-language/instrument-control/i2c-bus-specifications):

For reference, shielded 22 AWG twisted pair cables have capacitance in the range of 100-240 pF/m. So the maximum bus length of an I2C link is about 1 meter at 100 Kbaud, or 10 meters at 10 Kbaud. Unshielded cable typically has much less capacitance, but should only be used within an otherwise shielded enclosure. 

>> ...parameter called i2c_gpio_delay_us...

>I know for a fact based on the slave device MCU datasheet that the i2c is expected to run at 100kHz. Is there a range of delays I should try?

Set it to 4 (~50kHz) to see what happens. If it still doesn't work set it to 8 (~25kHz.)

>Additional information: Also, I have noticed that sometimes tiny spikes on the SDA signal occurs when i analyze the signal at higher speeds. What could cause this issue?

I don't know.

aidv, 2019-05-01:
This seems to be the problem:
[image]

Look at the end of the second byte (C0) we see a random spike on the SDA signal. That's the issue for sure I don't know why it happens, but that absolutely has to be the issue.

> When the I2C device is powered and without pullup resistors connected, measure the voltage between GND on the Raspberry Pi and SDA/SCL on the I2C device.

The voltage is close to 5v. 6v to be specific.

> Were bi-directional logic level shifters used...?

I am still unsure. I know the manufacturer specifies i2c compatibility. So I assume yes, but I won't take it for granted.

> GND/0V on the Raspberry Pi and GND/0V on the other system need to be connected to each other.

Does not change anything.

> Set it to 4 (~50kHz) to see what happens. If it still doesn't work set it to 8 (~25kHz.)

Tried both, still same.

aidv, 2019-05-02:
### Possible success

I cam across [the following thread in the Arduino forum](http://forum.arduino.cc/index.php?topic=595064.msg4090415#msg4090415) which I think solved my problem.

The user wothke says

> PS: A friend just told me that back in his electronics days they had once had a batch of defective EEPROMs that just acted like the one I've got now (at the time the problem had finally been recognized by the EEPROM producer and the complete batch was called back). Other than replacing the IC again, he suggested to maybe add some 10 or 100nF ceramic capacitor between GND/VCC close to the EEPROM to dampen disturbances that might trigger the problems of the flawed EEPROM.. I guess I'll just try that.

And later confirms by saying

> 100nF indeed seems to have solved the issue..

Bridging GND and 3v on the RPi with a 1uF cap seemed to have solved my problem. I still have to do further testing. Before I close this issue as it can be very useful information in the future and @fivdi might want to put this information in the README once I or someone else confirms this suspicion.

aidv, 2019-05-02:
I can with high certainty confirm that bridging the RPi GND and 3.3v with a 1uF capacitor fixed the incorrect readouts I was getting.

Closing and suggesting @fivdi to add this information somewhere in the repository.

closed 2019-05-02 by aidv

fivdi, 2019-05-02:
@aidv Thank you very much for the feedback. I'll add a comment to the readme describing the solution.

aidv, 2019-05-02:
@fivdi and thank YOU for your awesome support. I would recommend anyone that has the opportunity to test this solution further. This fixed the problem for me when using the hardware i2c bus. I will eventually also test the software i2c buses too.