fivdi/i2c-bus #74: Rpi with Arduino - writeSync question

ottogutierrez, 2019-08-14:
I'm doing communication between Rpi and an Arduino. I'm having trouble understanding the syntax of the bus.writeSync(addr,length,buffer).

When I send data to the Arduino, the arduino registers that I receive 2 pieces of data the length and the buffer. Is this intended? How would i send more than one byte? Is the buffer meant to be an array?

Thank you,

fivdi, 2019-08-14:
The buffer is a Node.js Buffer object. For example, if the address of the Arduino on the I2C bus is 0x23, and the 3 bytes 0x10, 0x11 and 0x12 should be written to the Arduino do the following:
```js
let buffer = Buffer.from([0x10, 0x11, 0x12]);

i2c1.i2cWriteSync(0x23, 3, buffer);
```

ottogutierrez, 2019-08-14:
Thanks for your clear explanation that makes sense to me now.

closed 2019-08-14 by ottogutierrez