fivdi/i2c-bus #76: deviceId throws exceptions for known active address

dkebler, 2019-09-10:
deviceId seems a new? available method so I decided to try it.

Just wondering if this method/function is only supported if the device itself supports it as when I try it for an existing device address (obtained via scan).  The returned callback error is

`Remote I/O error`

Not sure waht `Remote` means but maybe it refers to the i2c device being queried?  So maybe `I/O error` just means the  chip(mcp23008) doesn't support this function?   Maybe this is a newer i2c standard that device manufacturers are supposed to support?  I see nothing in the mcp23008 data sheet on this.

dkebler, 2019-09-10:
I see that error is number 121 which is a linux 
`#define EREMOTEIO       121     /* Remote I/O error */`
I kinda assumed this was an error you threw in the C code
Maybe the I/O refers to reading a sys file which is supposed to have that device info in it?

fivdi, 2019-09-11:
> deviceId seems a new? available method so I decided to try it.

It's fairly new. As mentioned in [History.md](https://github.com/fivdi/i2c-bus/blob/master/History.md#320--apr-21-2018) `deviceId` was added with i2c-bus@3.2.0.

> Just wondering if this method/function is only supported if the device itself supports it as when I try it for an existing device address (obtained via scan). The returned callback error is
>
>Remote I/O error

`deviceId` will only work if the I2C device supports it. For further information please take a look at the datasheet for the I2C device in question. Devices are not required to support `deviceId`.

The `Remote I/O error` comes from the I2C device driver. It this case the driver will have asked the I2C device to sends its `deviceId` but the device did not respond to the decide driver.

This isn't a problem with i2c-bus so I'll go ahead and close this issue.

closed 2019-09-11 by fivdi