fivdi/i2c-bus #81: Improve documentation

fivdi, 2019-09-26:
* Mention that bytes are unsigned integers in the range 0 to 255
* Mention that words are unsigned integers in the range 0 to 65535
* Mention that buffers are instances of [Buffer](https://nodejs.org/dist/latest/docs/api/buffer.html#buffer_buffer)

jscaglia, 2019-09-27:
> * Mention that buffers are instances of [Buffer](https://nodejs.org/dist/latest/docs/api/buffer.html#buffer_buffer)

Hi, a question.
If I want to save the word "test" on my NFC card.

I want to use the following methods (I don't know if it's okay to use these)

0x24 is my board address.
7 is the length I want to read

bus.i2cReadSync (0x24, 7, buffer?)
bus.i2cWrite (0x24, 7, buffer ?, cb?)

But I don't know what buffer is and cb ... will they have any example to pass me?

Greetings and thanks

fivdi, 2019-09-27:
@jscaglia please open a new issue to discuss your problem,

PR #84 (2019-10-03): Improve documentation
Fixes #81
closed 2019-10-03 by fivdi