fivdi/i2c-bus #83: Read and Write in NFC card

jscaglia, 2019-09-27:
Hi, a question.
If I want to save the word "test" on my NFC card.

I want to use the following methods (I don't know if it's okay to use these)

0x24 is my board address.
7 is the length I want to read

bus.i2cReadSync (0x24, 7, buffer?)
bus.i2cWrite (0x24, 7, buffer ?, cb?)

But I don't know what buffer is and cb ... will they have any example to pass me?

Greetings and thanks

fivdi, 2019-09-27:
> But I don't know what buffer is and cb ... will they have any example to pass me?

`buffer` is a Node.js [Buffer](https://nodejs.org/dist/latest/docs/api/buffer.html) object. Unfortunately this isn't mentioned in the documentation but #81 was created to improve the documentation.

`cb` is shorthand for `callback`. Callbacks are mentioned in the documentation, for example, in the description for class `Bus` [here](https://github.com/fivdi/i2c-bus#class-bus). The documentation for `i2cWrite` also mentions that `cb` is a completion callback [here](https://github.com/fivdi/i2c-bus#busi2cwriteaddr-length-buffer-cb).

If you're new to Node.js and not yet familiar with callbacks and asynchronous programming I'd suggest reading up on the topic, for example, [here](https://github.com/getify/You-Dont-Know-JS/blob/2nd-ed/async-performance/ch2.md).

I'm not familiar with using I2C to read NFC cards, but below are two example programs that show how to use i2cRead/i2cWrite and i2cReadSync/i2cWriteSync.

I'd suggest using either synchronous methods or asynchronous, but not both at the same time. The synchronous methods are always named with the suffix `Sync`,

#### A synchronous program that uses Buffers, i2cReadSync and i2cWriteSync to determine the temperature with a MCP9808 I2C temperature sensor
```js
const i2c = require('i2c-bus');

const MCP9808_ADDR = 0x18;
const TEMP_REG = 0x05;

const toCelsius = rawData => {
  let celsius = (rawData & 0x0fff) / 16;
  if (rawData & 0x1000) {
    celsius -= 256;
  }
  return celsius;
};

const i2c1 = i2c.openSync(1);

const writeBuf = Buffer.alloc(1);
writeBuf[0] = TEMP_REG;
i2c1.i2cWriteSync(MCP9808_ADDR, writeBuf.length, writeBuf);

const readBuf = Buffer.alloc(2);
i2c1.i2cReadSync(MCP9808_ADDR, readBuf.length, readBuf);

console.log(toCelsius(readBuf.readUInt16BE(0)));

i2c1.closeSync();
```
#### An asynchronous program that uses Buffers, completion callbacks, i2cRead and i2cWrite to determine the temperature with a MCP9808 I2C temperature sensor
```js
const i2c = require('i2c-bus');

const MCP9808_ADDR = 0x18;
const TEMP_REG = 0x05;

const toCelsius = rawData => {
  let celsius = (rawData & 0x0fff) / 16;
  if (rawData & 0x1000) {
    celsius -= 256;
  }
  return celsius;
};

const i2c1 = i2c.open(1, (err) => {
  if (err) throw err;

  const writeBuf = Buffer.alloc(1);
  writeBuf[0] = TEMP_REG;
  i2c1.i2cWrite(MCP9808_ADDR, writeBuf.length, writeBuf, (err) => {
    if (err) throw err;

    const readBuf = Buffer.alloc(2);
    i2c1.i2cRead(MCP9808_ADDR, readBuf.length, readBuf, () => {
      if (err) throw err;

      console.log(toCelsius(readBuf.readUInt16BE(0)));

      i2c1.closeSync(err => {
        if (err) throw err;
      });
    });
  });
});
```

fivdi, 2019-09-30:
@jscaglia Hopefully the above information was useful. I'll go ahead and close this issue as there doesn't appear to be a todo for i2c-bus.

closed 2019-09-30 by fivdi