fivdi/i2c-bus #88: Laster managing I2C-bus Event ?

yboujraf, 2019-12-27:
Dear @imyller , @fivdi , @munyirik  , @johntalton 

Are you managing i2c events ?

That means if any I2C device needs to share data, automatically hte MAster will catch the information and process it ?

Best Wishes for 2020
Youssef

fivdi, 2019-12-27:
@yboujraf Which type of I2C events from which type of I2C device are you referring too?

And best wishes for 2020 to you too.

fivdi, 2019-12-27:
One additional question. The title of this issue is "Laster managing I2C-bus Event ?". What does "Laster" mean?

yboujraf, 2019-12-27:
Dear @fivdi 

I have an PCF8574 defined as GPI.

And the PCF8574 is connected to my raspberry i2c bus with many other devices.

I don't want to poll every x ms the status of each i2c equipment, I would like to be informed for each change by "on-event-change" or something like this.

Because after, my raspberry forward this information to an automation in Real time.

Same issue with A/D converter, I have a fader (10kOhms), and I need to read in real time the data but ideally by event only.

Best Regards,
Youssef

fivdi, 2019-12-27:
>I have an PCF8574 defined as GPI.
>
>And the PCF8574 is connected to my raspberry i2c bus with many other devices.
>
>I don't want to poll every x ms the status of each i2c equipment, I would like to be informed for each change by "on-event-change" or something like this.

Page 11 of the [PCF8574 datasheet](http://www.ti.com/lit/ds/symlink/pcf8574.pdf) states the following:

- The PCF8574 device provides an open-drain output (INT) that can be connected to the interrupt input of a microcontroller. An interrupt is generated by any rising or falling edge of the port inputs in the input mode.

It should be possible to connect the open-drain INT output of the PCF8574 to a GPIO on a Raspberry Pi and detect the interrupts using the [onoff module](https://github.com/fivdi/onoff). The onoff `watch` method can detect hardware interrupts without polling. Note that the INT output on the PCF8574 is an open-drain output so a pull-up resistor will be needed. For an example of a Node.js I2C device driver that detects interrupts triggered by an I2C device take a look at [mcp9808-temperature-sensor](https://github.com/fivdi/mcp9808-temperature-sensor). An MCP9808 temperature sensor has an open-drain Alert output that serves a similar purpose as the open-drain INT output of the PCF8574. Keep the watch handler as short as possible to avoid missing interrupts.

>Same issue with A/D converter, I have a fader (10kOhms), and I need to read in real time the data but ideally by event only.

You'll need to look at the datasheet to see if the device supports interrupts.

fivdi, 2019-12-27:
I think the required information has been provided so I'll go ahead this issue.

closed 2019-12-27 by fivdi

yboujraf, 2019-12-27:
Thanks You are a great man.