fivdi/i2c-bus #89: i2c1.i2cRead Error in Promise function

OraCodeGroup, 2019-12-29:
This i2cRead runs correctly on Raspberry Pi:

i2c.openPromisified(1)
  // I2C Write Cmd
  .then( i2c1 => i2c1.i2cWrite(PN532_I2C_ADDRESS, (i2c_rwbuf[3] + PN532_FRAME_OVERHEAD - 1), i2c_rwbuf) 
    // I2C read ACK reply
    .then( _ => i2c1.i2cRead(PN532_I2C_ADDRESS, i2c_ackbuf.length, i2c_ackbuf))

But this i2cRead returns all zeros:

  i2c.openPromisified(1)
  // I2C Write Cmd
  .then( i2c1 => i2c1.i2cWrite(PN532_I2C_ADDRESS, (i2c_rwbuf[3] + PN532_FRAME_OVERHEAD - 1), i2c_rwbuf) 

    // I2C read ACK reply
    .then( _ => {
		i2c1.i2cRead(PN532_I2C_ADDRESS, i2c_ackbuf.length, i2c_ackbuf) 
	})

Why does putting the i2cRead into a function cause the read to fail?

fivdi, 2019-12-29:
> Why does putting the i2cRead into a function cause the read to fail?

Because the function doesn't have a `return` statement so it returns undefined.

fivdi, 2019-12-29:
In other words, it should be
```js
    .then( _ => {
		return i2c1.i2cRead(PN532_I2C_ADDRESS, i2c_ackbuf.length, i2c_ackbuf) 
	})
```

fivdi, 2019-12-29:
I'll go ahead and close this as it's not an i2c-bus issue.

closed 2019-12-29 by fivdi