fivdi/i2c-bus #90: checkByte error on correct byte values

VaderMester, 2020-04-02:
I'm using this lib in my Homebridge plugin.
I'm getting checkbyte errors.
Below byteToPass value is the byte my function is passing to i2c, I write it out in console to see if I'm doing something wrong.
```
[4/2/2020, 10:27:13 AM] [Pi Thermostat] Initializing Thermostat accessory...
byteToPass: 254
[4/2/2020, 10:27:13 AM] Error: Invalid byte 254
    at checkByte (/home/pi/shared/homebridge-pi-thermostat_Zone1-master/node_modules/i2c-bus/i2c-bus.js:105:11)
    at Bus.writeByte (/home/pi/shared/homebridge-pi-thermostat_Zone1-master/node_modules/i2c-bus/i2c-bus.js:406:5)
```

As you can see the Byte I'm passing is correct. It first got the error with a value 255, and tried it with 254. No idea what can be the issue.
Init of the bus is done like this:
 `   this.i2c1 = i2c.openSync(1);        //open /dev/i2c-1`
Function call:
  ```
MCP23017.prototype._send = function (cmd, values) {
    this.i2c1.writeByte(this.address, cmd, values, function (err) {
      if (err) {
        console.error(err);
      }
    });
  };
```

fivdi, 2020-04-02:
Is the value being passed to `writeByte` an unsigned integer?

In other words, does `Number.isInteger(values)` return true?

If not, that would explain the problem.

VaderMester, 2020-04-02:
Yep, you were right.
It is strange, because all of my variables are declared in hex, like
```
var REGISTER_GPIOA = 0x00,
  REGISTER_GPIOB = 0x01,
  REGISTER_GPIOA_PULLUP = 0x0C,
  REGISTER_GPIOB_PULLUP = 0x0D,
  READ_GPIOA_ADDR = 0x12,
  READ_GPIOB_ADDR = 0x13,
  WRITE_GPIOA_ADDR = 0x14,
  WRITE_GPIOB_ADDR = 0x15;
```

Putting this into my calling functions brute-force solved the issue :)

`
.
.
      var reg = (cmd & 0xff);
      var data = (values & 0xff);
      i2c1.writeByte(this.address, reg, data, function (err) {
.
.
`

fivdi, 2020-04-02:
Thank you for providing the feedback and good to hear that it's working now.

closed 2020-04-02 by fivdi