fivdi/i2c-bus #91: am2322 of temperature moisture sensor

raspberrypi-duke, 2020-04-18:
Hello,
I use am2322 of temperature moisture sensor. Is there a way using wake-up and the weight by the async function? When wake-up is done by the lead function(receiveByte()), a small error goes out to a server console. 
Thank you. 
duke

fivdi, 2020-04-18:
I'm not familiar with the am2322 so I may not be able to provide much help.

What error message is displayed on the console by the program posted above?

raspberrypi-duke, 2020-04-18:
Hi,

Thank you very much for your reply.

am2322 datasheet
https://www.isweek.com/Uploads/20180528/5b0bbbc09cff2.pdf

error display
{ [Error: , Remote I/O error] errno: 121, code: ‘’, syscall: ‘receiveByte’ }

thanks,
duke :)

fivdi, 2020-04-18:
> Device address is 0xb8.
> I used for a raspberry pi, so it shifts 1 bit(device address is 0x5c).

Yes, this is the correct thing to do.

The am2322 appears to have very specific and unusual timing requirements. 

There is a sample Python program on the following page that reads from an am2322:

https://www.denshi.club/pc/raspi/5raspberry-pi-zeroiot26-4-am2322.html

The Python program call `read_byte` (which is the same as `receiveByte` in JavaScript) and expects an error. When the error occurs, it then read the temperature and humidity.

You could try to convert the Python program to JavaScript to see if it works.

raspberrypi-duke, 2020-04-18:
Hi, fivdi.

Thank you very much for your best software.

Is there a way to close the i2c which opened in Wake-up? When not closing i2c and reading a sensor many times, will it be lack of a memory?

thanks,
duke :)

fivdi, 2020-04-18:
I see that the code in the first comment has completely changed. Please don't change code like this. It will make it difficult or impossible for people reading the issue later to understand everything. Post the modified code in a new comment instead.

Does the code now function correctly?

The code should be re-structured to use only one `PromisifiedBus` object , not two. Then it will not be necessary to close the first one. The code should also be modified to prevent the "wake-up" and "read the sensor" code from running at the same time which is currently the case.

raspberrypi-duke, 2020-04-18:
Hi, fivdi.

Cord change, I’m sorry. An error goes out to this cord in wake, a sensor can be read. I wanted to know whether it'll be lack of a memory by not closing i2c of wake.

thanks,
duke :)

fivdi, 2020-04-18:
Yes, not closing the bus object will cause file descriptor leaks.

You should use catch to ignore the error.

raspberrypi-duke, 2020-04-18:
Hi, fivdi.

It can't close by this cord.
Could you instruct the best way?

// wake-up the sensor
i2c.openPromisified(1).
then(i2c1 => i2c1.receiveByte(addr).
   then(_ => {
      return new Promise((resolve) => {
         setTimeout(() => {
            resolve(i2c1.close());
         }, 1);
      })
   })
).
catch(i2c1 => {
   i2c1.close();
   console.log;
});

thanks,
duke :)

fivdi, 2020-04-18:
Please take a look at a Promises tutorial.

closed 2020-04-18 by raspberrypi-duke