fivdi/i2c-bus #93: License question

nielsnl68, 2020-05-26:
Hello @fivdi 

a couple of years ago i created a Node red lib depending on your lib, .
Today i found out that my lib was said it as under Apache 2 license on NPM and on github i made it GPL. Wanting to modify the license to the right one,  i checked yours lib and found out you are using the MIT license. And now i am wondering of you have any preferences to what license the lib i made should have. so i can update it asap. 

thanks for your great work.

fivdi, 2020-05-26:
> And now i am wondering of you have any preferences to what license the lib i made should have.

I have no preference to what license the lib you made should have.

nielsnl68, 2020-05-26:
Okay thanks for your answer, i will make it Apache 2.

fivdi, 2020-05-26:
Ok

closed 2020-05-26 by fivdi