fivdi/i2c-bus #94: Please update your code to address deprecated Buffer() call use

dachshund-digital, 2020-07-08:
Please update your code to address deprecated Buffer() call use.  Following warning now results when using your module.

(node:14197) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.

fivdi, 2020-07-08:
There are no deprecated usages of `Buffer()` in the current i2c-bus codebase or at least I can't find any.
What version of i2c-bus are you using? Perhaps there were deprecated usages of `Buffer()` in older versions of i2c-bus.
Can you provide code that can be used to reproduce the issue?

fivdi, 2020-07-12:
As mentioned above there are no deprecated usages of `Buffer()` in the current i2c-bus codebase so I'll go ahead and close the issue.

closed 2020-07-12 by fivdi