fivdi/i2c-bus #95: Error: ENXIO, No such device or address on writeByteSync

YindSoft, 2020-07-08:
I'm having a strange error in that function:

`(node:3168) UnhandledPromiseRejectionWarning: Error: ENXIO, No such device or address
    at Bus.writeByteSync (/root/domoticize/node_modules/i2c-bus/i2c-bus.js:411:9)
    at SI1132.reset (/root/domoticize/si1132.js:179:19)
    at SI1132.initialize (/root/domoticize/si1132.js:129:12)
    at SI1132.init (/root/domoticize/si1132.js:96:7)
`

But if I put a console.log in the function, it works.. and doesn't give any error.
Any idea?

I added this to the start of the function to check the data.
`console.log(`${addr} ${cmd} ${byte}`);`

If I leave that console.log it wont throw an error.. maybe it's some sync issue here.

fivdi, 2020-07-08:
If adding a small amount of code that logs to the console works-around the issue then maybe it's a timing issue. I'm not familiar with the [Si1132 sensor](https://www.silabs.com/documents/public/data-sheets/Si1132.pdf) but section 4.2 Command Protocol of the datasheet does indicate that the sensor has special timing constraints. Does your code fulfill all the timing requirements specified in the datasheet?

YindSoft, 2020-07-08:
> If adding a small amount of code that logs to the console works-around the issue then maybe it's a timing issue. I'm not familiar with the [Si1132 [...]

Yes, I thinks its a timing issue. It says: "No Commands should be issued to the device for at least 1 ms after a Reset is
issued."
and I'm just sending everything sync so it might be faster than that.

I will make a little delay there. Thanks!

fivdi, 2020-07-12:
This doesn't appear to be an i2c-bus problem so I'll go ahead and close the issus.

closed 2020-07-12 by fivdi